stereolabs/zed-ros-wrapper #602: Unsupported encoding when running zed laserscan nodelet on

BraedenCu, 2020-07-28:
**Describe the bug**
unsupportig Encoding (mono16) error occurs when running the zed laserscan nodelet. 
.
**Device information:**
 - OS: Ubuntu
 - OS version:18.04
 - CPU:i7
 - GPU Nvidia: RTX 2060
 - ZED SDK Version:v3.1.2
 - ROS Wrapper version:latest master

**Expected behavior**
[ERROR] [1595943781.355664934]: Could not convert depth image to laserscan: Depth image has unsupported encoding: mono16

**Additional context**
This issue has only manifested itself after I completely reinstalled ros and all associated packages.
My other desktop did not have this problem, with the only difference being the zed sdk version.

Myzhar, 2020-07-28:
@BraedenCu "OpenNI" mode (mono16) was uncorrectly the default mode for depth image, I fixed the problem setting it to false.
Let me know if this solved the issue

BraedenCu, 2020-07-28:
That corrected the issue. Thank you for the extremely quick response.

closed 2020-07-28 by BraedenCu