stereolabs/zed-ros-wrapper #604: [Question] publish with different fequency(rate)

as0027111, 2020-07-31:
Hi, I am trying to apply zed2 with jetson TX2 to implement self-driving.
In my test, using the example of "position tracking" and "object detection" individually, it can get about 30 and 10 fps in the order.(test with `time` library)
But when I use zed-ros-wrapper to subscribe this two topics, 
`/zed2/zed_node/pose` and `/zed2/zed_node/obj_det/objects` 
it always warning 
`Elaboration takes longer (0.103736 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.`
seem like the object detection is the bottleneck?

how could I get the position with  30 fps and detection result with 10 fps separately?

In addition, is there any approach to get faster result with TX2 than 10 fps, or is this frequency normal?

Myzhar, 2020-07-31:
Hi @as0027111 
Object Detection is an high consuming task and the TX2 can't reach better performances.
The problem is that currently you cannot get the two topics at different rates because object detection uses the "synchronous" mode by default.
The ZED SDK allow to perform Object Detection asynchronously, but this feature is not yet available in the ROS wrapper.
I will add it in the backlog since it's really useful.

as0027111, 2020-07-31:
Thanks for your answering, it sounds not good.
I will try to implement this feature by two way.
Case one, I finish detection ,and publish position and result of detection asynchronously. I think it needs the ideal about multi-thread. How could I publish asynchronously? Is there any code to refer, I just a rookie of ROS and OS, sorry.
Case two, I try to turn on the function of "object detection" of ZED2, and use the published image to do object detection by my own (use `jetson-interference` or `YOLO`). Is the rate of publishing image faster than 10 fps? And could this way be accomplish?

Myzhar, 2020-08-24:
@as0027111 currently the ROS wrapper works in a synchronous way, so image publishing follows the rate of the detection process. We are working on adding the asynchronous detection to unlink the two processes and getting the best from each of them.

Myzhar, 2020-08-31:
Available with https://github.com/stereolabs/zed-ros-wrapper/commit/2ab8e863008ea17f282fa30310d142a13859ce49
Will be merged soon into master

as0027111, 2020-09-18:
Hi, thanks for your working.
I have already used the SDK to grab the information of sensor and detection asynchronously, and send them by socket scheme.
I will try the newest version later, thanks you.

closed 2020-09-30 by Myzhar