stereolabs/zed-ros-wrapper #606: Lens model with Kalibr for Zed2

FaboNo, 2020-08-04:
Hi There

I am doing a calibration with Kalibr and I am following th instructions provided in the page:
https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-
I used this line (I changed the zed with zed2)
`$ kalibr_calibrate_cameras --bag Kalibr_data.bag --topics /zed/zed_node/left/image_rect_color /zed/zed_node/right/image_rect_color --models pinhole-radtan pinhole-radtan --target april_grid.yaml`

But I got the following warning from Kalibr:
`[ WARN] [1596570850.291301]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)`

Therefore my question is do the pinhole-radtan model is still valid for the Zed2 or do I have to use another pinhole model?

Thank you for you help

Regards

Fabrice

Myzhar, 2020-08-04:
Hi @FaboNo 
The lenses of the ZED2 have a large FOV, not like a fisheye, but large.
Try to use the model suggested in this Kalibr issue 
https://github.com/ethz-asl/kalibr/issues/298
Please let me know if it works

FaboNo, 2020-08-04:
@Myzhar thank you for your answer, I still have the issue. But i think that I have to create a new bag with smaller images. Indeed the size of the calibration target (grid) is really small with respect to the size of the images so I will probably need to reduce the size of the images. I will let you know.

Myzhar, 2020-08-24:
@FaboNo can you please update the ticket or close it if you found a solution?

FaboNo, 2020-08-24:
@Myzhar I am sorry - I forgot to close it

closed 2020-08-24 by FaboNo

alexmillane, 2021-09-03:
@FaboNo I'm having the same issue, what was the solution in the end. Thanks so much :)

Aishkrish18, 2023-08-10:
I just have a question regarding the IMU update rate of ZED2, is it 360 Hz or 400 when you echo the imu topic? In zed2 the rate is 360 hz ,and i dont know if this is correct ?

ManChrys, 2023-10-05:
hello @Aishkrish18 

i have the same issue. did you find anything ?

Thanks

Myzhar, 2023-10-05:
Please do not reply to closed issues.
Report bugs and feature requests on new issues eventually tagging old closed issues if relevant, or create a thread on the [Stereolabs forum](https://community.stereolabs.com/) for questions.

Aishkrish18, 2023-10-05:
> hello @Aishkrish18
> 
> i have the same issue. did you find anything ?
> 
> Thanks

Hey, I just changed the frequency to 200 hz since it works fine for my application, i changed it in the common.yaml file. So at the moment it works well for me. I am not sure what  application you are using it for. But if you want to use it in 400 hz then I'm sorry i don't know the solution .

ManChrys, 2023-10-05:
@Aishkrish18 

#606