stereolabs/zed-ros-wrapper #607: IMU values difference betwwen zed_wrapper and zed_open_capture[Question]

FaboNo, 2020-08-05:
Hi,

Sorry to bother you with another question but we would like to unerstand why there are such differences between IMU values coming from the zed_wrapper:
```
header: 
  seq: 5353
  stamp: 
    secs: 1596616271
    nsecs:  70961989
  frame_id: "zed2_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00151527953502
  y: -0.000940317278931
  z: -0.000119983220765
angular_velocity_covariance: [6.049236592215545e-10, 0.0, 0.0, 0.0, 6.299619364771527e-10, 0.0, 0.0, -0.0, 5.826384766534753e-10]
linear_acceleration: 
  x: -0.367738366127
  y: 0.0641402825713
  z: 9.79715538025
linear_acceleration_covariance: [0.008283733390271664, 0.0, 0.0, 0.0, 0.005874685477465391, 0.0, 0.0, -0.0, 0.006347923539578915]

```
and the ones coming from zed_open_capture:

```
**** New IMU data ****
 * Timestamp: 1596616361967024364 nsec
 * Frequency: 401.547 Hz
 * Accelerations [m/s²]: -9.89321 -0.266088 0.529779
 * Angular Velocities [°/s]: 0.244141 -0.183105 0.0305176
**** New IMU data ****

```

This is significative for the angular velocities. 

Thank you very much for your support and kindness

Myzhar, 2020-08-05:
Hi @FaboNo 
the sensor values published by the ROS wrapper are compensated by the SDK using the factory calibration values, instead the ZED Open Capture driver emits the raw values as emitted by the sensors, performing only a simple rescaling to match the measure units

tftangming, 2020-08-17:
So what version is more useful?

obraun-sl, 2020-08-17:
It depends on what you need. If you want to get fused quaternion from IMU data, then it's better to use the ZED SDK. 
If you don't want to handle ZED SDK dependencies (CUDA) and want to have a low level access to camera data (as the sdk provides), then it's better to use the open capture driver.

Myzhar, 2020-08-24:
@FaboNo can you please update the ticket or close it if you found a solution to your question?

FaboNo, 2020-08-24:
@Myzhar  is it possible to extract the EKF from the SDK to fuse raw IMU data? It might be interesting in case we want do not want to handle CUDA dependancies as mentioned by @obraun-sl

Myzhar, 2020-08-24:
Hi @FaboNo 
the EKF inside the SDK uses CUDA because it fuses visual information with inertial information. 
Furthermore it is tuned on factory calibrated IMU values and it cannot be used with RAW IMU data.
For ROS I normally use the EKF or even better the UKF provided by the `robot_localization` package: 
http://docs.ros.org/melodic/api/robot_localization/html/index.html

FaboNo, 2020-08-24:
Hi @Myzhar thank you for your answer, so assuming that  we use the open capture API only - I fully agree that we are not leveraging on the full capabilities of ZED2 - it is not really possible to use the  "raw/uncalibrated" IMU data and the left/right images with a VIO algorithm?

Myzhar, 2020-08-24:
The best way to use them is to run a tool like Kalibr to get the correct calibration parameters:
https://github.com/ethz-asl/kalibr/wiki
RAW sensors data are affected by drifts, mainly for the gyro sensor, that will produced highly instable results, even if filtered with an EKF.

FaboNo, 2020-08-24:
@Myzhar thanks

closed 2020-08-24 by FaboNo

mandelyoni, 2023-04-10:
Dear sir,
What's the best way to get raw IMU data measurements through ROS?
The ROS topic  /zed2i/zed_node/imu/data_ raw seems compensated.
Thanks