stereolabs/zed-ros-wrapper #608: [BUG] move_base unable to access pointcloud

BraedenCu, 2020-08-06:
**Describe the bug**
 Client [/move_base] wants topic /zed/zed_nodelet/point_cloud/cloud_registered to have datatype/md5sum [sensor_msgs/PointCloud/d8e9c3f5afbdd8a130fd1d2763945fca], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.

Device information:
    OS: Ubuntu
    OS version:18.04
    CPU:i7
    GPU Nvidia: RTX 2060
    ZED SDK Version:v3.1.2
    ROS Wrapper version:latest master

**Additional context**
This error occurs when attempting to use the move_base node with the ros navigation stack.

Myzhar, 2020-08-06:
Hi @BraedenCu 
this is not a bug of the ROS wrapper. The `move_base` node subscribes to the "old" pointcloud topic, while the ZED ROS wrapper published `pointcloud2` topics.
Pointcloud was declared as deprecated, but it has never been removed because many nodes continue to use it.
You can read more here: https://answers.ros.org/question/12046/pointcloud-vs-pointcloud2-for-unstructured-point-clouds/

The solution is changing the parameters of your navigation stack in order to use pointcloud2 instead of pointcloud (local planner? global planner?) or (not recommended) to use a node to convert pointcloud2 to pointcloud: http://wiki.ros.org/point_cloud_converter

Myzhar, 2020-08-24:
@BraedenCu can you please update the ticket or close it if you found a solution?

closed 2020-08-31 by Myzhar