stereolabs/zed-ros-wrapper #610: IMU orientation_covariance reports nan values!

saeedarabi92, 2020-08-10:
Hi all, 

When I echo `/zed2/zed_node/imu/data` I receive `nan` values for `orientation_covariance`. I am using Jetson TX2 and cloned master branch of `zed_ros_wrapper`. What am I missing? Any help is appreciated. 

```
nvidia@nvidia-desktop:~/Desktop/Aria_navigation$ roslaunch zed_wrapper zed2.launch
... logging to /home/nvidia/.ros/log/e88b234e-db16-11ea-944a-00044bc7359a/roslaunch-nvidia-desktop-10668.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:36293/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: True
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 15
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path:
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed2/zed_node/pos_tracking/floor_alignment: False
 * /zed2/zed_node/pos_tracking/imu_fusion: True
 * /zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed2/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed2/zed_node/pos_tracking/map_frame: map
 * /zed2/zed_node/pos_tracking/odometry_frame: odom
 * /zed2/zed_node/pos_tracking/path_max_count: -1
 * /zed2/zed_node/pos_tracking/path_pub_rate: 2.0
[... 155 lines cut ...]
nvidia@nvidia-desktop:~$ rostopic echo -n1 /zed2/zed_node/imu/data
header:
  seq: 1058
  stamp:
    secs: 1597070725
    nsecs: 562746771
  frame_id: "zed2_imu_link"
orientation:
  x: -0.00422687735409
  y: -0.0121427755803
  z: 0.0
  w: 0.999917328358
orientation_covariance: [nan, nan, nan, nan, nan, nan, nan, nan, nan]
angular_velocity:
  x: 0.000468096607612
  y: 0.00215668967772
  z: -0.00194725389847
angular_velocity_covariance: [-6.511824237941176e-10, 0.0, 0.0, 0.0, 8.310105044895048e-10, 0.0, 0.0, -0.0, 7.032450657581959e-10]
linear_acceleration:
  x: 0.246236264706
  y: -0.0779369324446
  z: 9.787566185
linear_acceleration_covariance: [-0.009481723420321941, 0.0, 0.0, 0.0, 0.007963255047798157, 0.0, 0.0, -0.0, 0.009519890882074833]
---
```

obraun-sl, 2020-08-10:
Hi,
What version of the ZED SDK are you using?
There was a bug on IMU covariance fixed on 3.2.0 . See here : 
https://www.stereolabs.com/developers/release/

Make sure you have at least that version.

saeedarabi92, 2020-08-10:
I am using version 3.1.0.
Lemme try 3.2.0. and I'll update you.

saeedarabi92, 2020-08-10:
Reinstalling the SDK solved the problem. Thank you.
Closing this.

closed 2020-08-10 by saeedarabi92