stereolabs/zed-ros-wrapper #611: Questions on camera-imu transformation

hanzheteng, 2020-08-20:
Good evening,

After reading through all the available docs and looking into code stack, I am still confused about this. The goal is to obtain the transformation matrices between IMU and cameras (i.e. extrinsic parameter). 

I notice that during roslaunch, the Camera-IMU Transform is available in the following form, as you mentioned in https://github.com/stereolabs/zed-ros-wrapper/issues/597#issuecomment-663461646.
```
[ INFO] [1595584347.430331450]: Camera-IMU Transform: 
 5596CB1354C0
0.999626 0.025546 0.009792 -0.002000
-0.025588 0.999664 0.004221 -0.023000
-0.009681 -0.004470 0.999943 -0.002000
0.000000 0.000000 0.000000 1.00000
```
Can you please specify what is the reference frame? I would assume this is the transform of `imu_link` represented in `camera_center` frame, because `camera_center` frame is the parent frame of `imu_link` according to tf tree and [the documentation.](https://www.stereolabs.com/docs/ros/zed-node/#transform-frames) 

Also, there is a rostopic named `left_cam_imu_transform` that will publish the same transformation as the above one (at least the translation part is the same if not all). Then I would assume the reference frame is `left_camera_frame`, which is not consistent with my understanding above. (The current description on the documentation page is a bit vague to me.)

My third question is about the fixed or floating property of `imu_link` frame, following the discussion in issue https://github.com/stereolabs/zed-ros-wrapper/issues/384. When I launch RViz and play with Zed camera, I see that the `imu_link` moves according to my manipulation of the camera but the `camera_center` or `base_link` frame remains static, meaning that the transformation between camera and imu is not fixed but changing all the time. If so, what should we specify in the extrinsic parameter (which is expected to be a fixed transform)? Ideally my expectation is that `camera_center` and `imu_link` will move together but keep a fixed transform.

Thank you for your time and patience.

Myzhar, 2020-08-24:
Hi @hanzheteng 

> Can you please specify what is the reference frame? I would assume this is the transform of imu_link represented in camera_center frame, because camera_center frame is the parent frame of imu_link according to tf tree and the documentation.

the transform is refered to the left camera sensor as you write here:

> Also, there is a rostopic named left_cam_imu_transform that will publish the same transformation as the above one (at least the translation part is the same if not all). Then I would assume the [...]

> My third question is about the fixed or floating property of imu_link frame, following the discussion in issue #384. When I launch RViz and play with Zed camera, I see that the imu_link moves [...]

What you see in Rviz is the TF in "world frame". The IMU frame changes with higher speed respect to the "base_link" frame, that's why you see is updating FASTER than the all the other frames.
But this does not mean that the IMU <-> CAMERA transform changes in time, it is always the same in local frame coordinates.
Rviz is a good tool to see with eyes the status of a robotic systems, but it is not a good tool to really understand the evolution of the system itself when the system evolves faster than Rviz can run.

jorgemia, 2020-08-24:
I'm also trying to understand the IMU link a bit better. Would it be correct to add it into my urdf like this, relative to the left camera frame (based on what is published in left_cam_imu_transform)?

```
<joint name="${ camera_name}_imu_link_joint" type="fixed">
     <parent link="${ camera_name}_left_camera_frame"/>
     <child link="${ camera_name}_imu_link"/>
     <origin xyz="-0.002 -0.023 -0.002" rpy="0.00138 -0.00116 -0.00176" />
</joint>

<link name="${ camera_name}_imu_link" />
```

Myzhar, 2020-08-24:
Yes, that's correct if you use that URDF with only that camera. We cannot add that fixed link to thedefault URDF because it is different for each camera, but you can do it if you want. 
Remember to modify the source code of the nodelet disabling the publishing of the TF, otherwise you will have problems. Simply comment this part of code and recompile:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L2895-L2909

jorgemia, 2020-08-25:
> We cannot add that fixed link to the default URDF because it is different for each camera, but you can do it if you want.

@Myzhar is it different because the location varies slightly on each model?

Myzhar, 2020-08-25:
Exactly... and it is different in ZED2 and ZED-M

hanzheteng, 2020-09-01:
Thanks for the clarification. What @jorgemia suggested also works in my case. We do need a fixed transform (in the tf tree) between camera and imu. 

> that's why you see is updating FASTER than the all the other frames.
> it is not a good tool to really understand the evolution of the system itself when the system evolves faster than Rviz can run.

I don't quite understand why the publishing speed can affect visualization results. If I run tf_echo for `imu_link` with respect to `left_camera_frame` , it will show me the same conclusion that this transform is changing as I move the camera. If I subscribe to the `left_cam_imu_transform` topic, it will also show me that this transform is changing. 

From the data published in imu topic, I can infer that this data is w.r.t. a fixed frame attached to imu. The transform can also be obtained by kalibr I believe. To this end, I am not sure what is the meaning of the floating frame currently attached to the imu frame (i.e. how to intuitively/physically interpret this floating frame). 

Lastly, I think the floating joint option is ideal for applications like robot arm, where we can assign a rosnode to publish this real-time tf when needed (I see zed is currently following this design). However, in our case, we need to attach a fixed frame to imu, such that acc and gyro can publish their data w.r.t. this fixed imu frame, otherwise I am not entirely sure how to interpret the data from imu topic. I can understand that leaving it as a floating joint would make urdf file more concise (in a single file, expanded by macro), since zedm and zed2 have different displacement for imu, but would it be possible to have two urdf files for zedm and zed2, respectively, with fixed joint description? I think this might be more clear for development. 

Please kindly let me know if I missed any important aspect in the design.

Myzhar, 2020-09-01:
> I don't quite understand why the publishing speed can affect visualization results. If I run tf_echo for `imu_link` with respect to `left_camera_frame` , it will show me the same conclusion that [...]

Think about the camera TF published at 15 Hz (camera set to 15FPS), the IMU TF is published at 400Hz, so there are 26 IMU TFs looking in the future respect to the last camera TF. That's why Rviz does not know what to do with these 26 TFs.

The IMU frame can also be static, you can modify the URDF adding the transform from the SDK. We use a floating frame because we must get the transform from the SDK when the node is just started and static TF from URDF are just published.

Your argumentation are all correct and are perfectly suited for a well known robot configuration. You can modify the ZED node and the URDF to match your configuration, that's the good of open source software.

There is another problem with IMU and TFs: the pose published by the IMU is w.r.t., like you say, but the frame of the IMU is relative to the position of the left camera, so the pose must be transformed before publishing it...

hanzheteng, 2020-09-02:
> we must get the transform from the SDK when the node is just started and static TF from URDF are just published.
> PS The IMU camera transform is not the same for all the ZED Mini and for all the ZED2

Understood. To achieve more accurate results for each specific camera, this transform must be obtained when the node is started. 

> Think about the camera TF published at 15 Hz (camera set to 15FPS), the IMU TF is published at 400Hz, so there are 26 IMU TFs looking in the future respect to the last camera TF. That's why Rviz does not know what to do with these 26 TFs.

I still don't see the problem of the missing 26 TFs and I don't think this is the fault by Rviz, because `tf_echo` command line request will give me the same result. In my understanding, `tf_echo` or Rviz will just show me the latest calculated camera-imu transform, based on the latest TFs it received, respectively, from`left_camera_frame` and `left_camera_optical_frame` (15Hz), and from `left_camera_frame` and `imu_link` (400Hz). The excess TFs will be discarded if not used. This is my expectation of the behavior.

Overall, based on your explanation, I guess the following understanding might be correct. 

Recall that the behavior in Rviz is 1) camera is fixed to the ground/world all the time, and 2) imu_link is moving as I rotate the camera. 

Once the system starts, the first/initial transform is the actual transform used between camera and imu. The following acc and gyro data will be published w.r.t. this frame. As I rotate camera, the system will estimate the pose of the imu, so that it shows in Rviz the pose change w.r.t. the initial frame. Since the joint between camera and imu is floating, so there is no constraint to force camera frame move together with imu pose. Instead, the camera is still rigidly attached to the ground since the transform between camera frame and world frame (or odom/map, the frame Rviz is referring to) is fixed according to the URDF description.

Please kindly let me know if any part is misunderstood.

jorgemia, 2020-09-22:
@Myzhar regarding the point you said above about disabling the publishing of TF

> Remember to modify the source code of the nodelet disabling the publishing of the TF, otherwise you will have problems. Simply comment this part of code and recompile:
> https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L2895-L2909

With the latest fix from #627 we can simply disable the tf using the `sensors/publish_imu_tf` parameter right? (Need to update with the latest stuff)

Myzhar, 2020-09-22:
Hi @jorgemia 
that's correct, if you add the static TF to your URDF file than you must disable IMU TF publishing setting `sensors/publish_imu_tf` to false.

Myzhar, 2021-05-17:
Can you please update the status of this issue and close it if it's solved?

hanzheteng, 2021-05-18:
Thanks for the reminder. Will conduct an experiment and get back to this soon.

anath93, 2023-07-07:
@Myzhar I have a question regarding the same topic, did not feel like opening a new topic but if you suggest I can. I am trying to find 'rpy' to provide to my urdf. Already got the 'xyz' (-0.002 -0.023 0.00) coordinates using logger by enabling publish_tf and echo. Any thoughts on this ?

[image]

[image]

thanks 
Ani

Myzhar, 2023-07-07:
Hi @anath93 
please publish your question to the Stereolabs community forum.
GitHub issues should be used only for feature requests and bur reporting.