stereolabs/zed-ros-wrapper #615: [BUG] fatal error: zed_interfaces/RGBDSensors.h: No such file or directory

520xyxyzq, 2020-08-26:
**Describe the bug**
Error when catkin build zed-ros-wrapper : 
```
fatal error: zed_interfaces/RGBDSensors.h: No such file or directory
 #include "zed_interfaces/RGBDSensors.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
**Device information:**
 - OS: Ubuntu
 - OS version:  18.04
 - CPU: i7
 - GPU Nvidia:  RTX 3000
 - ZED SDK Version: v3.2.2
 - ROS Wrapper version: latest master

Myzhar, 2020-08-27:
Hi @520xyxyzq 
can you give me more information? Can you copy and paste all the compilation log?

520xyxyzq, 2020-08-27:
```
In file included from ~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_demux_nodelet/src/rgbd_sensor_demux.cpp:21:0:
~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_demux_nodelet/include/rgbd_sensor_demux.hpp:36:10: fatal error: zed_interfaces/RGBDSensors.h: No such file or directory
 #include "zed_interfaces/RGBDSensors.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_demux_nodelet/src/rgbd_sensor_demux.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp:21:0:
~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_sync_nodelet/include/rgbd_sensor_sync.hpp:40:10: fatal error: zed_interfaces/RGBDSensors.h: No such file or directory
 #include "zed_interfaces/RGBDSensors.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp.o] Error 1
In file included from ~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:21:0:
~/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:43:10: fatal error: zed_interfaces/reset_tracking.h: No such file or directory
 #include <zed_interfaces/reset_tracking.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/ZEDNodelets.dir/all] Error 2
make: *** [all] Error 2
cd ~/catkin_ws/build/zed_nodelets; catkin build --get-env zed_nodelets | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

Myzhar, 2020-08-27:
No, it's not because of the SDK version, maybe that you updated the wrapper, but you did not clear the cmake cache.
Please delete the folders `build` and `devel` in your catkin workspace, then try to compile again.

520xyxyzq, 2020-08-27:
This is the first time I compile this wrapper.

Myzhar, 2020-08-27:
OK, maybe that there is some circular dependency issue. I check it

Myzhar, 2020-08-27:
@520xyxyzq I just checked with a new catkin workspace, I cloned the repository and started `catkin_make` that reached the end of compilation without any problem.

Are you using a particular procedure to compile the wrapper different from what is suggested [here](https://github.com/stereolabs/zed-ros-wrapper#build-the-program)?

520xyxyzq, 2020-08-27:
Thank you Myzhar. I forgot to resdep install. It's working now

closed 2020-08-27 by 520xyxyzq

bulbage, 2020-09-10:
Hello,
I am getting this error but only under certain condition. I will explain.
I bought a zedmini and followed your instructions to install the sdk and associated ROS examples on two platforms:
my Dell workstation and Jetson NX. In both cases I am able to compile and run everything just great!

However, I also followed these instructions https://ardupilot.org/dev/docs/ros-cartographer-slam.html for the specific application I am trying to eventually build/integrate. Not there yet...

When I added those instructions I started getting this error:
```
catkin build 

--------------------------------------------------------- 

Profile:                     default 

Extending:          [cached] /opt/ros/melodic 

Workspace:                   /home/u1/catkin_ws 

--------------------------------------------------------- 

Build Space:        [exists] /home/u1/catkin_ws/build 

Devel Space:        [exists] /home/u1/catkin_ws/devel 

Install Space:      [unused] /home/u1/catkin_ws/install 

Log Space:          [exists] /home/u1/catkin_ws/logs 

Source Space:       [exists] /home/u1/catkin_ws/src 

DESTDIR:            [unused] None 

--------------------------------------------------------- 

Devel Space Layout:          linked 

Install Space Layout:        None 

--------------------------------------------------------- 

Additional CMake Args:       None 

Additional Make Args:        None 

Additional catkin Make Args: None 

Internal Make Job Server:    True 

Cache Job Environments:      False 

--------------------------------------------------------- 

Whitelisted Packages:        None 

Blacklisted Packages:        None 

--------------------------------------------------------- 

Workspace configuration appears valid. 

--------------------------------------------------------- 

[build] Found '11' packages in 0.0 seconds.                                                                                                                                                                 

[build] Updating package table.                                                                                                                                                                             

Starting  >>> cartographer_ros_msgs                                                                                                                                                                         

Starting  >>> ceres-solver                                                                                                                                                                                  
[... 146 lines cut ...]

sudo vi node.launch  

 

cd ~/catkin_ws/ 

sudo apt-get install ros-melodic-catkin python-catkin-tools   

sudo apt-get install python-catkin-tools   

catkin build 

 

Then added zed-ros-examples zed-ros-wrapper to src 

catkin build 
```

These exact steps work well on my workstation and not on NX. 
Any insight on how to proceed with this problem would be greatly appreciated.

Thank you.
Nick