stereolabs/zed-ros-wrapper #616: [Question] pointclouds holes and noise fix

tianshiz, 2020-08-27:
my camera is pointed 30degrees from the groundplane so there's a tilted view. Are the gaping holes in the pointcloud caused by the light reflection on the surface?

There;s quite a bit of noise going on here. Is there anything that I can do to improve it? I'm using all the default common.yaml settings.
camera view
[image]

front pointcloud view:
[image]
side view:
[image]

Myzhar, 2020-08-27:
Hi @tianshiz 
if you are using a ZED that does not have an IMU sensor the SDK does not have any information about the gravity vector, so it's normal that the view is tilted. You can set this information modifying the launch file:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed.launch#L35

To improve depth estimation you can use the ULTRA depth mode, simply set "depth/quality" to 3:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L40

As you say the gaping holes in the point cloud are caused by the light reflections on the surface.

Myzhar, 2021-05-17:
Can you please update the status of this issue and close it if it's solved?

closed 2022-04-19 by github-actions[bot]