stereolabs/zed-ros-wrapper #619: [Question] Camera-IMU transform broadcaster

bjsowa, 2020-09-02:
If the transform between camera and imu frames is static (does not change over time), why not use the `StaticTransformBroadcaster` to publish a latched message on the `/tf_static` topic, instead of publishing a lot of messages on the `/tf` topic?
Am I misunderstanding something?

Myzhar, 2020-09-02:
Hi @bjsowa 
this can be the right solution indeed also related to the ticket https://github.com/stereolabs/zed-ros-wrapper/issues/611
Furthermore I believe that it will solve also all the problems with Rviz.

Testing it in the `fix_imu_duplicated_ts` branch

Myzhar, 2020-09-02:
@bjsowa I tried your approach using `tf2_ros::StaticTransformBroadcaster`, but the TF is not published as it's not visible in the TF tree.

Do you notice something wrong with this code?

```
void ZEDWrapperNodelet::publishStaticImuFrame() {
    // Publish IMU TF as static TF
    if( !mPublishImuTf ) {
        return;
    }

    if(mStaticImuFramePublished) {
        return;
    }

    tf2_ros::StaticTransformBroadcaster transformImuBroadcaster;

    geometry_msgs::TransformStamped transformStamped;
    transformStamped.header.stamp = ros::Time::now();
    transformStamped.header.frame_id = mLeftCamFrameId;
    transformStamped.child_frame_id = mImuFrameId;
    sl::Translation sl_tr = mSlCamImuTransf.getTranslation();
    transformStamped.transform.translation.x = sl_tr.x;
    transformStamped.transform.translation.y = sl_tr.y;
    transformStamped.transform.translation.z = sl_tr.z;
    sl::Orientation sl_or = mSlCamImuTransf.getOrientation();
    transformStamped.transform.rotation.x = sl_or.ox;
    transformStamped.transform.rotation.y = sl_or.oy;
    transformStamped.transform.rotation.z = sl_or.oz;
    transformStamped.transform.rotation.w = sl_or.ow;

    // Publish transformation
    transformImuBroadcaster.sendTransform(transformStamped);

    NODELET_INFO_STREAM("Published static transform '" << mImuFrameId << "' -> '" << mLeftCamFrameId << "'" );

    mStaticImuFramePublished = true;
}
```

bjsowa, 2020-09-02:
Is something published on the `/tf_static` topic?

The `sendTransform` method takes a const reference so maybe the transformStamped object needs to live outside the function call. Could you try making the transformStamped object a class member variable?

Myzhar, 2020-09-02:
Yes, that solved the problem.

Fix available with commit https://github.com/stereolabs/zed-ros-wrapper/commit/20215c409f212c06e314ae088527a23e1dec7ee7

Thank you for your suggestion, sometimes easier solution are harder to be thought

closed 2020-09-07 by bjsowa