stereolabs/zed-ros-wrapper #620: Running ORB-SLAM3 with ZED2 always appears error.

yike-chen, 2020-09-03:
I has run ORB-SLAM3 with Stereo+IMU. I set the 
```

```

zdegrh, 2020-09-03:
你好，
我感觉相机到IMU的转换有问题，我也在惆怅这个事情。
在EuRoC数据集中，相机的前向是z，右向是x，下向是y，IMU标架在相机坐标系下的旋转矩阵是[[0,-1,0],[1,0,0],[0,0,1]]，配置文件是符合这样的定义的。
但是在zed中，相机标架定义不变的情况下，IMU标架在相机坐标系下的旋转矩阵是[[0,0,1],[-1,0,0],[0,-1,0]]，并不是/zed2/zed_node/left_cam_imu_transform给出的[[1,0,0],[0,1,0],[0,0,1]]。
可惜，想的挺好，实际做出来还是不行。
不知道问题出在哪里。。。

yike-chen, 2020-09-03:
> 你好，
> 我感觉相机到IMU的转换有问题，我也在惆怅这个事情。
> 在EuRoC数据集中，相机的前向是z，右向是x，下向是y，IMU标架在相机坐标系下的旋转矩阵是[[0,-1,0],[1,0,0],[0,0,1]]，配置文件是符合这样的定义的。
> 但是在zed中，相机标架定义不变的情况下，IMU标架在相机坐标系下的旋转矩阵是[[0,0,1],[-1,0,0],[0,-1,0]]，并不是/zed2/zed_node/left_cam_imu_transform给出的[[1,0,0],[0,1,0],[0,0,1]]。
> 可惜，想的挺好，实际做出来还是不行。
> 不知道问题出在哪里。。。

你用kalibr自己标定了结果吗？我自己标定了一个结果，把这个相机到IMU的转换关系放上去还是不行。 不知道问题出在哪里，是IMU频率太高？

zdegrh, 2020-09-03:
> 你用kalibr自己标定了结果吗？我自己标定了一个结果，把这个相机到IMU的转换关系放上去还是不行。 不知道问题出在哪里，是IMU频率太高？

没有标定。
一直报init extractor last KF is empty!
我试过改相机分辨率和fps，没试过改imu频率。我试试看。

yike-chen, 2020-09-03:
> 没有标定。
> 一直报init extractor last KF is empty!
> 我试过改相机分辨率和fps，没试过改imu频率。我试试看。

请问您这边可以运行了吗？麻烦您这边能运行的话告知我一下哈。

Myzhar, 2020-09-03:
@yike-chen @zdegrh can you please use English language?

yike-chen, 2020-09-03:
> @yike-chen @zdegrh can you please use English language?

Sorry. I will translate it into English.

yike-chen, 2020-09-03:
> 你好，
> 我感觉相机到IMU的转换有问题，我也在惆怅这个事情。
> 在EuRoC数据集中，相机的前向是z，右向是x，下向是y，IMU标架在相机坐标系下的旋转矩阵是[[0,-1,0],[1,0,0],[0,0,1]]，配置文件是符合这样的定义的。
> 但是在zed中，相机标架定义不变的情况下，IMU标架在相机坐标系下的旋转矩阵是[[0,0,1],[-1,0,0],[0,-1,0]]，并不是/zed2/zed_node/left_cam_imu_transform给出的[[1,0,0],[0,1,0],[0,0,1]]。
> 可惜，想的挺好，实际做出来还是不行。
> 不知道问题出在哪里。。。

English Version:
He thinks the Camera-IMU transform that causes the problem. In the EuRoC dataset, the camera front orientationis z, the right orientation is x, and the under orientation is y. The rotation matrix  of IMU framework in coordinate system of camera is [[0,-1,0],[1,0,0],[0,0,1]].
When the definition of camera frame remains unchanged in ZED,  the rotation matrix  of IMU framework in coordinate system of camera is [[0,0,1],[-1,0,0],[0,-1,0]], and it's not / zed2 / zed_ node/left_ cam_ imu_ [1,0,0], [0,1,0], [0,0,1]. 
He didn't know what the problem was. Me too.

yike-chen, 2020-09-03:
> 你用kalibr自己标定了结果吗？我自己标定了一个结果，把这个相机到IMU的转换关系放上去还是不行。 不知道问题出在哪里，是IMU频率太高？

English version:
Did you calibrate the results yourself with Kalibr? I calibrated a result by myself, but it remains do not works well with the new parameters. I don't know what the problem is. Is the frequency of IMU too high?

Myzhar, 2020-09-03:
The camera/IMU transform is in the standard default frame: X Forward, Y left, Z Top.
Looking at the first post I noticed this: `not IMU meas`
Have you subscribed to the right topic?

yike-chen, 2020-09-03:
> 没有标定。
> 一直报init extractor last KF is empty!
> 我试过改相机分辨率和fps，没试过改imu频率。我试试看。

He do not calibrate it.
Error is always appears "Init extractor last KF is empty!"
I've tried to change the camera resolution and FPS, but not the IMU frequency. I'll try.

Myzhar, 2020-09-03:
You cannot change the IMU frequency. It always runs at maximum speed: 400Hz.
Unless you activate the camera/IMU synchronization, in this case the IMU data will be published together with the releative image frame

yike-chen, 2020-09-03:
> The camera/IMU transform is in the standard ROS frame: X Forward, Y left, Z Top.
> Looking at the first post I noticed this: `not IMU meas`
> Have you subscribed to the right topic?
Yes, I have subcribed to the right topic with  **/zed2/zed_node/right/right/image_rect_color**. I have used it with the Stereo Mode, and it works well. Once I fused it with the IMU data, it doesn`t work.

Myzhar, 2020-09-03:
I was referring to the right imu topic, not the image topic

yike-chen, 2020-09-03:
> I was referring to the right imu topic, not the image topic

Sorry, I don`t  find the right imu topic. I only find these topics as following.
```
/zed2/joint_states
/zed2/zed_node/atm_press
/zed2/zed_node/confidence/confidence_map
/zed2/zed_node/depth/camera_info
/zed2/zed_node/depth/depth_registered
/zed2/zed_node/depth/depth_registered/compressed
/zed2/zed_node/depth/depth_registered/compressed/parameter_descriptions
/zed2/zed_node/depth/depth_registered/compressed/parameter_updates
/zed2/zed_node/depth/depth_registered/compressedDepth
/zed2/zed_node/depth/depth_registered/compressedDepth/parameter_descriptions
/zed2/zed_node/depth/depth_registered/compressedDepth/parameter_updates
/zed2/zed_node/depth/depth_registered/theora
/zed2/zed_node/depth/depth_registered/theora/parameter_descriptions
/zed2/zed_node/depth/depth_registered/theora/parameter_updates
/zed2/zed_node/disparity/disparity_image
/zed2/zed_node/imu/data
/zed2/zed_node/imu/data_raw
/zed2/zed_node/imu/mag
/zed2/zed_node/left/camera_info
/zed2/zed_node/left/image_rect_color
/zed2/zed_node/left/image_rect_color/compressed
/zed2/zed_node/left/image_rect_color/compressed/parameter_descriptions
/zed2/zed_node/left/image_rect_color/compressed/parameter_updates
/zed2/zed_node/left/image_rect_color/compressedDepth
/zed2/zed_node/left/image_rect_color/compressedDepth/parameter_descriptions
/zed2/zed_node/left/image_rect_color/compressedDepth/parameter_updates
/zed2/zed_node/left/image_rect_color/theora
/zed2/zed_node/left/image_rect_color/theora/parameter_descriptions
/zed2/zed_node/left/image_rect_color/theora/parameter_updates
/zed2/zed_node/left/image_rect_gray
/zed2/zed_node/left/image_rect_gray/compressed
/zed2/zed_node/left/image_rect_gray/compressed/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/compressed/parameter_updates
/zed2/zed_node/left/image_rect_gray/compressedDepth
/zed2/zed_node/left/image_rect_gray/compressedDepth/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/compressedDepth/parameter_updates
/zed2/zed_node/left/image_rect_gray/theora
/zed2/zed_node/left/image_rect_gray/theora/parameter_descriptions
/zed2/zed_node/left/image_rect_gray/theora/parameter_updates
/zed2/zed_node/left_cam_imu_transform
/zed2/zed_node/left_raw/camera_info
/zed2/zed_node/left_raw/image_raw_color
/zed2/zed_node/left_raw/image_raw_color/compressed
/zed2/zed_node/left_raw/image_raw_color/compressed/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_color/compressed/parameter_updates
/zed2/zed_node/left_raw/image_raw_color/compressedDepth
/zed2/zed_node/left_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_color/compressedDepth/parameter_updates
/zed2/zed_node/left_raw/image_raw_color/theora
/zed2/zed_node/left_raw/image_raw_color/theora/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_color/theora/parameter_updates
/zed2/zed_node/left_raw/image_raw_gray
/zed2/zed_node/left_raw/image_raw_gray/compressed
/zed2/zed_node/left_raw/image_raw_gray/compressed/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_gray/compressed/parameter_updates
/zed2/zed_node/left_raw/image_raw_gray/compressedDepth
/zed2/zed_node/left_raw/image_raw_gray/compressedDepth/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_gray/compressedDepth/parameter_updates
/zed2/zed_node/left_raw/image_raw_gray/theora
/zed2/zed_node/left_raw/image_raw_gray/theora/parameter_descriptions
/zed2/zed_node/left_raw/image_raw_gray/theora/parameter_updates
/zed2/zed_node/odom
/zed2/zed_node/parameter_descriptions
/zed2/zed_node/parameter_updates
/zed2/zed_node/path_map
/zed2/zed_node/path_odom
[... 86 lines cut ...]
/zed2/zed_node/right_raw/image_raw_gray/theora/parameter_updates
/zed2/zed_node/stereo/image_rect_color
/zed2/zed_node/stereo/image_rect_color/compressed
/zed2/zed_node/stereo/image_rect_color/compressed/parameter_descriptions
/zed2/zed_node/stereo/image_rect_color/compressed/parameter_updates
/zed2/zed_node/stereo/image_rect_color/compressedDepth
/zed2/zed_node/stereo/image_rect_color/compressedDepth/parameter_descriptions
/zed2/zed_node/stereo/image_rect_color/compressedDepth/parameter_updates
/zed2/zed_node/stereo/image_rect_color/theora
/zed2/zed_node/stereo/image_rect_color/theora/parameter_descriptions
/zed2/zed_node/stereo/image_rect_color/theora/parameter_updates
/zed2/zed_node/stereo_raw/image_raw_color
/zed2/zed_node/stereo_raw/image_raw_color/compressed
/zed2/zed_node/stereo_raw/image_raw_color/compressed/parameter_descriptions
/zed2/zed_node/stereo_raw/image_raw_color/compressed/parameter_updates
/zed2/zed_node/stereo_raw/image_raw_color/compressedDepth
/zed2/zed_node/stereo_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed2/zed_node/stereo_raw/image_raw_color/compressedDepth/parameter_updates
/zed2/zed_node/stereo_raw/image_raw_color/theora
/zed2/zed_node/stereo_raw/image_raw_color/theora/parameter_descriptions
/zed2/zed_node/stereo_raw/image_raw_color/theora/parameter_updates
/zed2/zed_node/temperature/imu
/zed2/zed_node/temperature/left
/zed2/zed_node/temperature/right
```

Myzhar, 2020-09-03:
This is the right (meaning correct) IMU topic:
/zed2/zed_node/imu/data

yike-chen, 2020-09-03:
> This is the right (meaning correct) IMU topic:
> /zed2/zed_node/imu/data

I have tried with **/zed2/zed_node/imu/data** or **/zed2/zed_node/imu/data*, but it does not work.

yike-chen, 2020-09-03:
Moreover, I found a problem that launching the ```roslaunch zed_display_rviz display_zed2.launch ```. The IMU status is sometimes abnormal.
[image]

Myzhar, 2020-09-03:
This is known, but it's a Rviz problem, not a IMU problem. The IMU TF runs at 400Hz, so the camera pose TF is always in late respect to it since it runs slower.
A possible solution is available in the `fix_imu_duplicated_ts` branch that will be merged in the master soon

yike-chen, 2020-09-03:
I found the same problem in [ORB-SLAM Issue 28](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/28). It seems the Calibration parameter resulting in the problem. I use the ZED2 factory parameters, why do I have these problems? I guess that the ZED SDK results in the problem.

yike-chen, 2020-09-04:
Currently [ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94) only supports Global Shutter cameras. So I can not use it to run the ORB-SLAM3.

closed 2020-09-04 by yike-chen

awenbigprawn, 2021-04-27:
Really? Why will a Global Shutter camera or rolling shutter camera make a difference? My image frames are actually stable but still get same problem of 
"PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map..."

MeisonP, 2021-07-23:
@yike-chen have you solved the problem of  orb_slam3 with ZED2  stereo-imu?

Geoffrey1500, 2021-08-14:
> I has run [ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3) with Stereo+IMU. I set the parameters following.
> 
> ```
> %YAML:1.0
> 
> #--------------------------------------------------------------------------------------------
> # Camera Parameters. Adjust them!
> #--------------------------------------------------------------------------------------------
> Camera.type: "PinHole"
> 
> # Camera calibration and distortion parameters (OpenCV) 
> # I get it from the topic of /zed2/zed_node/left/camera_info.
> Camera.fx: 537.50732421875
> Camera.fy: 537.50732421875
> Camera.cx: 630.62646484375
> Camera.cy: 362.08624267578125
> 
> # I use the rectification images, so set these with zeros.
> Camera.k1: 0.0
> Camera.k2: 0.0
> Camera.p1: 0.0
> Camera.p2: 0.0
> 
> Camera.width: 1280
> Camera.height: 720
> 
> # Camera frames per second 
> Camera.fps: 15.0
> 
> # IR projector baseline times fx (aprox.)
> # I get it from the topic of /zed2/zed_node/right/camera_info with P(4).
> Camera.bf: 64.62
> 
> # Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
> Camera.RGB: 1
> 
> # Close/Far threshold. Baseline times.
> ThDepth: 40.0
> 
> # Deptmap values factor
> DepthMapFactor: 1.0
> 
> # Transformation from camera 0 to body-frame (imu)
> # I get it from topic of /zed2/zed_node/left_cam_imu_transform, and it seems more precise with STARTER CONSOLE information "Camera-IMU Transform" 
> Tbc: !!opencv-matrix
>    rows: 4
>    cols: 4
>    dt: f
>    data: [0.999956756829, 0.00843152806614, 0.00392349541458, -0.00200000009499,
>           -0.0084334428508, 0.99996540322, 0.000471741896564, -0.02300000377,
>           -0.00391937794561, -0.000504810067849, 0.999992191792, -0.00200000009499,
>          0.0, 0.0, 0.0, 1.0]
> 
> # IMU noise
> # I get it from Project of **zed-examples/tutorials/tutorial 7 - sensor data/**.
> IMU.NoiseGyro: 0.007 # 1.6968e-04 
> IMU.NoiseAcc:  0.0016 # 2.0000e-3
> IMU.GyroWalk:  0.0019474 
> IMU.AccWalk:   0.0002509 # 3.0000e-3
> IMU.Frequency: 400
> ```
> 
> I have set all parametes with ZED2 SDK provided, but I can`t run well with ORB-SLAM3(Stereo+IMU). I run well with ORB-SLAM3(Stereo), once I fuse the IMU data it damaged. The error can be found following.
> 
> ```
> New Map created with 313 points
> build optimization graph
> start optimization
> end optimization
> update Keyframes velocities and biases
[... 180 lines cut ...]
> First KF:45; Map init KF:44
> New Map created with 255 points
> Fail to track local map!
> IMU is not or recently initialized. Reseting active map...
> LM: Active map reset recieved
> LM: Active map reset, waiting...
> LM: Reseting current map in Local Mapping...
> LM: End reseting Local Mapping...
> LM: Reset free the mutex
> LM: Active map reset, Done!!!
> mnFirstFrameId = 184
> mnInitialFrameId = 182
> 172 Frames set to lost
> not IMU meas
> First KF:46; Map init KF:45
> New Map created with 385 points
> Fail to track local map!
> IMU is not or recently initialized. Reseting active map...
> ```
> 
> Can you give me some advices to solve it? Or I must calibrate it with Kalibr?

你好，请问这个问题解决了嘛，我最近也是双目运行可以，但是双目加imu就报错，提示not enough accelerator

还请多多指教

awenbigprawn, 2021-08-15:
> 你好，请问这个问题解决了嘛，我最近也是双目运行可以，但是双目加imu就报错，提示not enough accelerator
>
> 还请多多指教

ZED2自带的IMU到摄像头的距离太近了，你看下你外参的文件里这一段：
# Transformation from camera 0 to body-frame (imu)
# I get it from topic of /zed2/zed_node/left_cam_imu_transform, and it seems more precise with STARTER CONSOLE information "Camera-IMU Transform" 
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0.999956756829, 0.00843152806614, 0.00392349541458, -0.00200000009499,
          -0.0084334428508, 0.99996540322, 0.000471741896564, -0.02300000377,
          -0.00391937794561, -0.000504810067849, 0.999992191792, -0.00200000009499,
         0.0, 0.0, 0.0, 1.0]
变换矩阵里平移项的长度都在毫米厘米级别，这对于SLAM算法来说近似于奇点一样，而且这也不是非常精确的值，你也不可能拆开zed2去量，一点点误差相对于2毫米的绝对距离都是很大的。我是通过在机器人上另一个地方固定了另一个IMU，然后输入绝对值稍微大一点的平移量，使用这个imu作为input，ORB就可以成功运行了。

提示not enough accelerator的原因：ORB-SLAM3的stereo+imu模式 初始化的时候，机器人不能停着不动，实际上你把机器人开动起来一段时间，ORB就会自动初始化成功了。

祝好。

RoZhong, 2021-12-03:
> **/zed2/zed_node/imu/data** or **/zed2/zed_node/imu/data_raw**

请问您这个问题解决了么

YuxuanSnow, 2022-03-23:
I think the problem should be the unstable IMU measurement. I print the timestamp (frequency) of each IMU measurement, and found that each IMU measurement has quite different frequency.

The formula to calculate the frequency is: `1 / (current_IMU_timestamp - last_IMU_timestamp)`.

[image]

kpmrozowski, 2022-04-19:
> I think the problem should be the unstable IMU measurement. I print the timestamp (frequency) of each IMU measurement, and found that each IMU measurement has quite different frequency.
> 
> The formula to calculate the frequency is: `1 / (current_IMU_timestamp - last_IMU_timestamp)`.

`dt` is being taken into account in ORB_SLAM3::IMU::Preintegrated::IntegrateNewMeasurement(const Eigen::Vector3f &acceleration, const Eigen::Vector3f &angVel, const float &dt).

Never the less the problem with `not IMU meas` still exists, has anyone solved this issue? @yike-chen @Myzhar ?

Veng97, 2022-06-02:
Has anyone solved this problem yet? I have ORB-SLAM3 working fine without IMU at 15Hz and even 30Hz. But i would much prefer to run at 30Hz + IMU if i can. Or at least 15Hz + IMU if i want to run many things at the same time.

# Transformation from camera 1 to camera 2 (left to right)
Stereo.T_c1_c2: !!opencv-matrix
  rows: 4
  cols: 4
  dt: f
  data: [1.0, 0.0, 0.0, 0.120065,
         0.0, 1.0, 0.0, 0.0,
        0.0, 0.0, 1.0, 0.0,
        0.0, 0.0, 0.0, 1.0]

# Transformation from camera 1 to body-frame (imu)
IMU.T_b_c1: !!opencv-matrix
  rows: 4
  cols: 4
  dt: f
  data: [0.0, 0.0, 1.0, 0.0,                                                                                                                                                                      
        -1.0, 0.0, 0.0, 0.0,                                                                                                                                                                       
        0.0, -1.0, 0.0, 0.0,                                                                                                                                                                      
        0.0, 0.0, 0.0, 1.0]

SKLotus, 2023-10-10:
请问解决了吗，我也是一直初始化不成功的问题