stereolabs/zed-ros-wrapper #623: [Question] Depth modes

oscarthorn, 2020-09-07:
Hi!

In the enum for depth modes (https://www.stereolabs.com/docs/api/group__Depth__group.html#ga8d542017c9b012a19a15d46be9b7fa43) there are 5 entries:
None, Performance, Qualtiy, Ultra, Last.
What happens if I set it to Last in the config file, does it just revert to Ultra? Or something else?

Thanks!

Myzhar, 2020-09-07:
Hi @oscarthorn 
LAST should not be used on any enum, it's internal and is undefined behavior, the same if you use a not allowed value in the config file.

dennisVi, 2020-09-25:
Hi Myzhar,

Could you please clarify the depth modes available, because there are conflicts in the documentation.

The yaml configuration file shows:   # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA

The SDK3 page you referenced also mentions these modes:
NONE | This mode does not compute any depth map. Only rectified stereo images will be available.
PERFORMANCE | Computation mode optimized for speed.
QUALITY | Computation mode designed for challenging areas with untextured surfaces.
ULTRA | Computation mode favorising edges and sharpness. Requires more GPU memory and computation power.
LAST

The actual SDK also shows these:
enum class DEPTH_MODE {
        NONE, /**< This mode does not compute any depth map. Only rectified stereo images will be available.*/
        PERFORMANCE, /**< Computation mode optimized for speed.*/
        QUALITY, /**< Computation mode designed for challenging areas with untextured surfaces.*/
        ULTRA, /**< Computation mode favorising edges and sharpness. Requires more GPU memory and computation power.*/
        LAST
    };

However, the ros-wrapper API shows:

PARAMETER | DESCRIPTION | VALUE
quality | Select depth map quality | 0: NONE, 1: PERFORMANCE, 2: MEDIUM, 3: QUALITY, 4: ULTRA

Normally, I would trust the source code and argue that the AI-based Ultra class must be '3'. However, I've ran multiple experiments and concluded that there are in fact 5 modes that give distinct results.
0: no depth is computed
1: performance mode
2: medium ? = does not work well on low-textured areas, but clearly distinct from 1 and 3.
3: quality ?  = improved performance on low-texture surfaces, but the confidence map is rather useless (either max or min confidence)
4: I think this is ultra ? ==> results resemble option 3 but with sharper edges and the confidence map is more realistic.

So quality 1 to 4 give distinct results. Also, setting the quality / depthmode to any value above 4 results seems to yield output as if quality 4 is used. 

I've searched the zed-wrapper source code to see if an additional quality level was added by means of post-processing, but did not find anything. Can you please elaborate on which settings are officially supported and why setting the quality to an (officially) unsupported '4' gives best results?

Best regards,
Dennis

Myzhar, 2020-09-25:
Hi @dennisVi 
there was a typo in the comment of the YAML file, it has been fixed many months ago. 
The correct values are those illustrated in the ZED documentation.
Indeed if you update the wrapper to the latest release of the master branch you can see that the typo has been fixed:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L40

closed 2021-05-17 by Myzhar

tugbakara, 2022-08-23:
Hi, @Myzhar I have a problem about depth modes, I have Nvidia Jetson AGX and ZED2, when I changed the depth mode 1 to 2 or 1 to 3 AGX is shutting down? Is ZED2 compatible with these modes or not? I get confused because in another issues with these modes can be used with ZED2.

Myzhar, 2022-08-23:
@tugbakara this is a closed issue. Please open a new issue with the new problem or send an email to support@stereolabs.com providing more information about your configuration