stereolabs/zed-ros-wrapper #625: [Question] catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_demux_nodelet/include/rgbd_sensor_demux.hpp:36:10: fatal error: zed_interfaces/RGBDSensors.h: No such file or directory   #include "zed_interfaces/RGBDSensors.h"

bulbage, 2020-09-10:
Hello,
I am getting this error but only under certain condition. I will explain.
I bought a zedmini and followed your instructions to install the sdk and associated ROS examples on two platforms:
my Dell workstation and Jetson NX. In both cases I am able to compile and run everything just great!

However, I also followed these instructions https://ardupilot.org/dev/docs/ros-cartographer-slam.html for the specific application I am trying to eventually build/integrate. Not there yet...

When I added those instructions I started getting this error:
```
catkin build 

--------------------------------------------------------- 

Profile:                     default 

Extending:          [cached] /opt/ros/melodic 

Workspace:                   /home/u1/catkin_ws 

--------------------------------------------------------- 

Build Space:        [exists] /home/u1/catkin_ws/build 

Devel Space:        [exists] /home/u1/catkin_ws/devel 

Install Space:      [unused] /home/u1/catkin_ws/install 

Log Space:          [exists] /home/u1/catkin_ws/logs 

Source Space:       [exists] /home/u1/catkin_ws/src 

DESTDIR:            [unused] None 

--------------------------------------------------------- 

Devel Space Layout:          linked 

Install Space Layout:        None 

--------------------------------------------------------- 

Additional CMake Args:       None 

Additional Make Args:        None 

Additional catkin Make Args: None 

Internal Make Job Server:    True 

Cache Job Environments:      False 

--------------------------------------------------------- 

Whitelisted Packages:        None 

Blacklisted Packages:        None 

--------------------------------------------------------- 

Workspace configuration appears valid. 

--------------------------------------------------------- 

[build] Found '11' packages in 0.0 seconds.                                                                                                                                                                 

[build] Updating package table.                                                                                                                                                                             

Starting  >>> cartographer_ros_msgs                                                                                                                                                                         

Starting  >>> ceres-solver                                                                                                                                                                                  
[... 146 lines cut ...]

sudo vi node.launch  

 

cd ~/catkin_ws/ 

sudo apt-get install ros-melodic-catkin python-catkin-tools   

sudo apt-get install python-catkin-tools   

catkin build 

 

Then added zed-ros-examples zed-ros-wrapper to src 

catkin build 
```

These exact steps work well on my workstation and not on NX. 
Any insight on how to proceed with this problem would be greatly appreciated.

Thank you.
Nick

sakthivelj, 2020-09-16:
Errors     << zed_nodelets:make /home/tyrebot/catkin_ws/logs/zed_nodelets/build.make.000.log                                                 
In file included from /home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_demux_nodelet/src/rgbd_sensor_demux.cpp:21:0:
/home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_demux_nodelet/include/rgbd_sensor_demux.hpp:36:10: fatal error: zed_interfaces/RGBDSensors.h: No such file or directory
 #include "zed_interfaces/RGBDSensors.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_demux_nodelet/src/rgbd_sensor_demux.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp:21:0:
/home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/rgbd_sensors_sync_nodelet/include/rgbd_sensor_sync.hpp:40:10: fatal error: zed_interfaces/RGBDSensors.h: No such file or directory
 #include "zed_interfaces/RGBDSensors.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp.o] Error 1
In file included from /home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:21:0:
/home/tyrebot/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:43:10: fatal error: zed_interfaces/reset_tracking.h: No such file or directory
 #include <zed_interfaces/reset_tracking.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/ZEDNodelets.dir/all] Error 2
make: *** [all] Error 2
cd /home/tyrebot/catkin_ws/build/zed_nodelets; catkin build --get-env zed_nodelets | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.............................................................................................................................................
Failed     << zed_nodelets:make                     [ Exited with code 2 ]                                                                   
Failed    <<< zed_nodelets                          [ 7.2 seconds ]                                                                          
Abandoned <<< zed_wrapper                           [ Unrelated job failed ]                                                                 
Finished  <<< turtlebot3_fake                       [ 8.4 seconds ]                                                                          
Finished  <<< turtlebot3_slam                       [ 3.9 seconds ]                                                                          
[build] Summary: 10 of 12 packages succeeded.                                                                                                
[build]   Ignored:   2 packages were skipped or are blacklisted.                                                                             
[build]   Warnings:  2 packages succeeded with warnings.                                                                                     
[build]   Abandoned: 1 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 20.8 seconds total.                                                                                                         
[build] Note: Workspace packages have changed, please re-source setup files to use them.

sakthivelj, 2020-09-16:
We are facing the same issue we are trying the all above step but I can't build the package.

Myzhar, 2020-09-16:
It's a dependency issue, but I cannot understand why it happens.
As temporary solution you can try this procedure:
1. Compile only `zed_interfaces`
`$ catkin_make --only-pkg-with-deps zed_interfaces`
2. Switch back to all packages
`$ catkin_make -DCATKIN_WHITELIST_PACKAGES=""`
3. compile everything else
`$ catkin_make`

sakthivelj, 2020-09-16:
tyrebot@tyrebot-PhotoGAUGE:~/catkin_ws$ catkin_make --only-pkg-with-deps zed_interfaces
Base path: /home/tyrebot/catkin_ws
Source space: /home/tyrebot/catkin_ws/src
Build space: /home/tyrebot/catkin_ws/build
Devel space: /home/tyrebot/catkin_ws/devel
Install space: /home/tyrebot/catkin_ws/install
Whitelisted packages: zed_interfaces
####
#### Running command: "cmake /home/tyrebot/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/tyrebot/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/tyrebot/catkin_ws/install -DCATKIN_WHITELIST_PACKAGES=zed_interfaces -G Unix Makefiles" in "/home/tyrebot/catkin_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python27; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:20 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python27

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  CMakeLists.txt:20 (find_package)

-- Using CATKIN_DEVEL_PREFIX: /home/tyrebot/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/tyrebot/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/tyrebot/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/tyrebot/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: zed_interfaces
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - zed_interfaces
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_interfaces'
-- ==> add_subdirectory(zed-ros-wrapper/zed_interfaces)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_interfaces: 3 messages, 13 services
-- Configuring incomplete, errors occurred!
See also "/home/tyrebot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/tyrebot/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

sakthivelj, 2020-09-16:
A building alone the **zed_interfaces** packages  also fail to build

Myzhar, 2020-09-16:
@sakthivelj  in your case I think that this is self explaining:
```
Could not find the following Boost libraries:

      boost_python27
```

LanceChen, 2020-10-13:
I'm running into the same issue as @bulbage . I'm running this on a Jetson TX2, using `catkin build` tools. The only package in the workspace is the zed ros wrapper. 

Building just zed_interfaces first as @Myzhar suggested did not work.

I also tried changing the CMakeLists for zed_nodelets. I inserted `REQUIRED` to the `find_package` call, like so:

`find_package(catkin REQUIRED COMPONENTS
    roscpp
    image_transport
    rosconsole
    sensor_msgs
    stereo_msgs
    std_msgs
    message_filters
    tf2_ros
    nodelet
    tf2_geometry_msgs
    message_generation
    diagnostic_updater
    dynamic_reconfigure
    zed_interfaces
)
`

I got the following error as a result:

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "zed_interfaces"
  with any of the following names:

    zed_interfacesConfig.cmake
    zed_interfaces-config.cmake

  Add the installation prefix of "zed_interfaces" to CMAKE_PREFIX_PATH or set
  "zed_interfaces_DIR" to a directory containing one of the above files.  If
  "zed_interfaces" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:44 (find_package)

Myzhar, 2020-10-13:
Try to remove `zed_interfaces` from the list of `find_package` and leave it only in the list of `catkin_package`. 
Sometimes this solves the issue

LanceChen, 2020-10-13:
Removing `zed_interfaces` from `find_package` produces the original error (could not find .h files). Seems to suggest that `find_package` is failing to include `zed_interfaces` to begin with?

Lost-Monkey-5, 2020-10-26:
@LanceChen is correct from what I was able to replicate. The line in zed_nodelets CMakeLists: 

`list(APPEND CMAKE_PREFIX_PATH "/opt/ros/$ENV{ROS_DISTRO}/")` 

seems to cause `find_package` to be unable to locate `zed_interfaces` for the errors described above when you build the program using: 

`catkin build zed_nodelets --cmake-args -DCMAKE_BUILD_TYPE=Release` 

I am able to build the package using:

 `catkin_make -DCMAKE_BUILD_TYPE=Release`

Myzhar, 2020-10-26:
@Lost-Monkey-5 that's a line to be able to use the CMake files in QtCreator. It can be removed if generates problems with other parts.

Let me know if removing it solves your problems

Lost-Monkey-5, 2020-10-26:
@Myzhar I am able to build all the zed packages without that line and no longer get the errors about missing header files using `catkin_make` or `catkin build` so that seems to solve the issue. @bulbage Should see if removing that from the CMakeLists will fix the issue,

@Myzhar why does `find_package(catkin COMPONENTS ...)` not have `find_package(catkin REQUIRED COMPONENTS ...)` in the CMakeList? The issue would have been easier to find that way.

Myzhar, 2020-10-26:
Good to know. I fix it in the next release that will be ready as soon as we solve the issue #629

oscarthorn, 2020-11-19:
Hi! Will this be fixed soon? I would be nice not having to edit files to build the package.

Myzhar, 2020-11-19:
@oscarthorn now it is fixed in the master branch

worldpotato, 2021-03-01:
I think this can be closed since the original cause is fixed @bulbage

closed 2021-03-01 by Myzhar