stereolabs/zed-ros-wrapper #630: zed_wrapper zed launch error

harisiva1, 2020-09-25:
Hello,
upon launching zed_wrapper zed i get the following error.can someone help?

[image]
[image]

Myzhar, 2020-09-25:
Hi @harisiva1 
it seems that the ZED nodelet cannot be launched because the library `libnvcuvid.so.1` cannot be loaded.
Is CUDA correctly installed in your system?

harisiva1, 2020-09-26:
Hello,
It was one of the nvidia driver issue and now it works.
Thanks!

Myzhar, 2020-09-26:
You're welcome. I close the issue

closed 2020-09-26 by Myzhar