stereolabs/zed-ros-wrapper #631: [Question] ASCII encoding problem when subscribed to a ZED image topic.

yogeshgajjar, 2020-09-29:
[image]

I'm facing this error when a ZED topic is subscribed. I'm working on a object tracker and I require the video feed. I created a simple program to subscribe the topic "/zed/zed_node/rgb/image_rect_color" and used cv_bridge in the callback function to convert ROS image message to OpenCV image. 

The syntax I used is as follows, 
`cv_image = bridge.imgmsg_to_cv2(data, desired_encoding='passthrough')`

I don't how to work around this problem. It would be great if someone help me solving this problem. 
Thanks!

Myzhar, 2020-09-30:
Hi @yogeshgajjar 
the ZED images are in bgra8 format, I guess that they are not recognized by the node that you are using.
Have you tried changing `desired_encoding='passthrough'` to `desired_encoding='bgr8' or similat?

yogeshgajjar, 2020-09-30:
Hi, @Myzhar Yes, I tried all possible encoding pattens. It gave the same error. I'm facing this issue with the python version. Turns out the C++ version is working. But, I want it to work on python as my entire object tracking stack is in Python.

Myzhar, 2020-10-01:
@yogeshgajjar there is a scheduled update to write a nodelet to convert native BGRA images from the ZED Wrapper to BGR just to avoid this kind of problems. I hope this will solve this kind of problems