stereolabs/zed-ros-wrapper #632: [ZED2 with Jetson NX and ros_openpose not working-dies after few seconds]

AswinkarthikeyenAK, 2020-09-29:
**Describe the bug**
The ros node dies few seconds after it is launched.
**Device information:**
 - CPU: [Jetson NX board]
 - ZED SDK Version: [latest sdk for jetpack 4.4]
 - ROS Wrapper version: [latest master]

**To Reproduce**
Install ros_openpose `https://github.com/ravijo/ros_openpose`.
Use your laptop or the desktop to be the ROS_Master and the NX to be the client. 
Run the ros master from the laptop/desktop and then launch the ros_openpose.
Please let me know if you cannot reproduce this error. Would be happy to give you more information

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

`[ INFO] [1601419508.570561931]: ZED connection -> SUCCESS
[ INFO] [1601419510.571108990]:  ...  ZED ready
[ INFO] [1601419510.571663901]: ZED SDK running on GPU #0

FO] [1601419510.823202505]: Advertised on topic /zed2/zed_node/atm_press @ 25 Hz
[ INFO] [1601419510.825647141]: Advertised on topic /zed2/zed_node/temperature/left @ 25 Hz
[ INFO] [1601419510.827926433]: Advertised on topic /zed2/zed_node/temperature/right @ 25 Hz
[ INFO] [1601419510.830235485]: Advertised on topic /zed2/zed_node/left_cam_imu_transform [LATCHED]
[ WARN] [1601419517.119825215]: Waiting for datum...
[ WARN] [1601419528.800092745]: Waiting for datum...
[ WARN] [1601419537.522834099]: Waiting for datum...
[ WARN] [1601419547.387181647]: Waiting for datum...
[ WARN] [1601419557.940109518]: Waiting for datum...
[ WARN] [1601419567.431675096]: Waiting for datum...
[ WARN] [1601419577.771493037]: Waiting for datum...
[ WARN] [1601419587.432623925]: Waiting for datum...

`

[logfiles.zip](https://github.com/stereolabs/zed-ros-wrapper/files/5302188/logfiles.zip)

Myzhar, 2020-09-30:
Hi @AswinkarthikeyenAK 
the warning `Waiting for datum...` make me suppose that something is wrong with the topics names.
Sometimes the CLI command `$ roswtf` executed while the nodes are running can give useful information

AswinkarthikeyenAK, 2020-09-30:
Hi @Myzhar 
Waiting for datum... comes from the ros_openpose.
It prints after 10ms when there is no colour image is received.
While previously using it with nano, the msg waiting for datum usually prints for like 10 times and then when the camera starts to capture the image, it goes of.

I will run $ roswtf and let you what happens.

Thanks for your reply

AswinkarthikeyenAK, 2020-09-30:
Hi @Myzhar , Please find the output of the roswtf.

Thanks

`$ roswtf`
````
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 4 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING Communication with node [/visualizer] is very slow
WARNING Communication with node [/echo] is very slow
WARNING The following node subscriptions are unconnected:
 * /zed2/zed2_state_publisher:
   * /zed2/joint_states
 * /rosOpenpose:
   * /zed2/zed_node/depth/depth_raw_registered

WARNING No tf messages
```

Myzhar, 2020-09-30:
This is important I guess:
```
 * /rosOpenpose:
   * /zed2/zed_node/depth/depth_raw_registered
```
Does OpenPose want a message in OpenNI format? (unsigned int 16 bit)

AswinkarthikeyenAK, 2020-09-30:
Yes. It is.
Also, I am running the ros_openpose wrapper that worked with the nano. 

I am happy to share more information if need. Please feel free to let me know if there is some other information you need

Myzhar, 2020-09-30:
OK, in this case be sure that the parameter `depth/openni_depth_mode` is set to `true` (see `common.yaml`).
You can also verify that the topic `/zed2/zed_node/depth/depth_raw_registered` is correctly published with the command
`$ rostopic list`

AswinkarthikeyenAK, 2020-09-30:
Hi, It is set to true already.
and rostopic list shows the topic.
When I can the program now, this was the error I received.

`REQUIRED process [zed2/zed_node-5] has died!
process has died [pid 13932, exit code -11, cmd /home/lams/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/lams/.ros/log/2897b0cc-02f1-11eb-a232-982cbc26877f/zed2-zed_node-5.log].
log file: /home/lams/.ros/log/2897b0cc-02f1-11eb-a232-982cbc26877f/zed2-zed_node-5*.log
Initiating shutdown!
`

Myzhar, 2020-09-30:
I just verified and if you are using the latest version of the wrapper from the master branch you must subscribe to `/zed2/zed_node/depth/depth_registered` (without `raw`).
We unified the topic name for both OpenNI and not OpenNI mode to create less problems and to follow the ROS standard.
I add this to the release note, sorry for the information missing.

AswinkarthikeyenAK, 2020-10-08:
Hi @Myzhar ,
Thanks for the information.
I have made the necessary changes, and then tried. It seemed to work fine openpose in ros, until recently. 
Please find the image below.
[image]
Initially it was link this,
[image]

I did not make any changes to the parameters of the zed2 camera files in the zed_ros_wrapper. 
I cannot really figure out what happened here
Can you kindly let me know if something needs to be changed,please?

Please let me know if you need more information.
Thanks

closed 2021-05-17 by Myzhar