stereolabs/zed-ros-wrapper #633: Jetson TX1 + Zed-m Drop WIFI link

fgastald, 2020-10-05:
Hello,
i am in trouble with the latest SDK for ZED and the latest JETPACK available.
As soon as i launch zedm.launch the wifi connection drops (it loose the connection to the AP). 
Killing the zed ros node will restore the wifi connection.
I have seen that others have the same issue.
jetson_clock is activated, on jtop i just can see a raise of processor utilization when zed ros node is launched but nothing crtitical.
Do you have any tips for me?
thanks in advance for the kindly help
Best regards
Fabio

Myzhar, 2020-10-05:
Hi @fgastald 
the wifi connection does not drops, but I suppose it's simply saturated.
What topics are you subscribing through wifi?

fgastald, 2020-10-05:
Thank you for the promtly reply :)
Actually it's a default configuration i just change to PERFORMANCE in the depth setting and nothing else.
The problem is that it loose the connection (if you look on top left the wifi is trying to reconnect).
By the way i'm not getting anything over wifi right now as i'm working directly on the jetson (no remote pc)

fgastald, 2020-10-05:
Hi Walter,
Just an update, today i tried to put a powered usb 3.0 hub instead of the non powered 3.0 hub i used before.
Now at the first test (i had few time to try but i can update you later) it seems that with the powered hub the wifi disconnection does not come up anymore.

I noticed by the way that after boot the Jetson did not connect to the ZED-M i needed to detach the usb cable from the camera and reattach it, that way it recognize the camera, i had this issue before with the unpowered hub and jetpack 3.2

Now with jetpack 4.2 and the unpowered hub seemed to be solved and the camera were recognized from the beginning after power up.

...so.... strange issues.

Will do some more tests and will let you know
best regards
Fabio

closed 2021-05-17 by Myzhar