stereolabs/zed-ros-wrapper #634: [Question] why I enabling object  detection in zed2 will cause zed_node-2 crash?

FishInWave, 2020-10-13:
problem

Every time I tried to start object detection in my zed2, no matter with "true" of od_enabled in zed2.yaml or `rosservice call start_object_detection` while running zed2 node, an unexpected crash occurs as: 
>  [ INFO] [1602570829.500964519]: *** Starting Positional Tracking ***
> [ INFO] [1602570829.501075503]:  * Waiting for valid static transformations...
> [ INFO] [1602570829.501242539]: Static transform Sensor to Base [zed2_left_camera_frame -> base_link]
> [ INFO] [1602570829.501309693]:  * Translation: {0.000,-0.060,0.000}
> [ INFO] [1602570829.501362683]:  * Rotation: {0.000,-0.000,0.000}
> [ INFO] [1602570829.501428290]: Static transform Sensor to Camera Center [zed2_left_camera_frame -> zed2_camera_center]
> [ INFO] [1602570829.501477870]:  * Translation: {0.000,-0.060,0.000}
> [ INFO] [1602570829.501531439]:  * Rotation: {0.000,-0.000,0.000}
> [ INFO] [1602570829.501594404]: Static transform Camera Center to Base [zed2_camera_center -> base_link]
> [ INFO] [1602570829.501638500]:  * Translation: {0.000,0.000,0.000}
> [ INFO] [1602570829.501691065]:  * Rotation: {0.000,-0.000,0.000}
> [ INFO] [1602570829.601848698]: Initial ZED left camera pose (ZED pos. tracking): 
> [ INFO] [1602570829.601986893]:  * T: [0,0.06,0]
> [ INFO] [1602570829.602086174]:  * Q: [0,0,0,1]
> [ INFO] [1602570829.717731662]: *** Starting Object Detection ***
> ================================================================================REQUIRED process [zed_node-2] has died!
> process has died [pid 14199, exit code -11, cmd /home/xyw/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/xyw/.ros/log/56036690-0d01-11eb-afa0-acd1b8168b47/zed_node-2.log].
> log file: /home/xyw/.ros/log/56036690-0d01-11eb-afa0-acd1b8168b47/zed_node-2*.log
> Initiating shutdown!
> ================================================================================
> [zed_node-2] killing on exit
> [zed2_state_publisher-1] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

Additionally, there is already the file named objects_1.3.model in /usr/local/zed/resources.

[ZED_Diagnostic_Results.txt](https://github.com/stereolabs/zed-ros-wrapper/files/5369510/ZED_Diagnostic_Results.txt)
I have also run ZED_Diagnostic tool and upload the result file. In 235 line of this file,  it says "unknown" about "InstalledZEDSDKVersion", so is it cause my crash and it is needed to be re-installed?

Thank you for any help.

Myzhar, 2020-10-13:
Hi @FishInWave 
I suggest you to re-install the SDK.
Let me know if this works

FishInWave, 2020-10-15:
> Hi @FishInWave
> I suggest you to re-install the SDK.
> Let me know if this works

It didn't work.
Besides, I have run od on another PC with 2080Ti, it just worked nomally.And its "ZED_Diagnostic_Results" also says  "unknown" about "InstalledZEDSDKVersion".
I guess the node crashes just because my laptop don't have enough GPU memory for initialization.

Myzhar, 2020-10-15:
Yes, maybe that's the problem. Have you tried to compile and run the OD sample outside ROS?
https://github.com/stereolabs/zed-examples/tree/master/object%20detection

FishInWave, 2020-10-21:
> Yes, maybe that's the problem. Have you tried to compile and run the OD sample outside ROS?
> https://github.com/stereolabs/zed-examples/tree/master/object%20detection

Yeah，I have tried this example （both cpp and py） on laptop and PC. IThey have the same issue.

closed 2020-10-22 by FishInWave