stereolabs/zed-ros-wrapper #635: [Question] Error when creating the depth map

Cram3r95, 2020-10-22:
Hi folks,

I am trying to run the depth map using:

cuda 9.0
ZED SDK 3.2.2
Latest version of zed-ros-wrapper

The RGB images run fine, but not the depth map. Main parameters are as following:

"
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic

depth:
    quality:                    1                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)
"

However, sometime it works, but if I move the camera, it suddenly stop working (in terms of depth, it works but publishing a black image). What can I do?

Myzhar, 2020-10-22:
Can you explain better how are you displaying the depth map?
It is a floating point image and if it is not correctly normalized before displaying it's easy to get a black image in case of short distances.
Please provide also a screenshot to better understand the problem

Cram3r95, 2020-10-22:
Sorry, I meant regarding RVIZ visualization. I think it is correctly calculated. The issue was that in RVIZ, if the parameter "Normalize_Range" is enable, it tries to calculate the min and max depth in order to normalize the colours. However, if you disable this parameter and set some fixed numbers, let's say 0.1 m and 20 m, it properly represents the depth map.

Now I have another question: Which is the min and max distance (with a high confidence) detected by the depth map? Does it depend on the zed position or tilt angle? 

We are using this depth map and the rectified left image in order to know the xyz position of a detection using Visual Object Tracking (using CenterNet + DeepSORT) for our Autonomous Vehicle architecture. Since it is visual object tracking, I want to know the 3D position.

Myzhar, 2020-10-22:
The depth map represents for each pixel its perpendicular distance from the camera sensor plane and it does not depends on camera position or orientation.
The minimum and maximum distances are set using the relative parameters with some limitations, as explained in the ZED SDK documentation:
https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#a115c93ba994ec7b5c853375d647a936e

Cram3r95, 2020-10-22:
@Myzhar I have just read the doc you sent. By default, the "float depth_minimum_distance" is 70 cm for ZED, what it is fine for our purposes. However, the maximum distance it is not illustrated here. Where can I check (and change if possible) these parameters? zed-ros-wrapper offers this option?

Cram3r95, 2020-10-22:
@Myzhar I just got it. It is in zed-ros-wrapper/zed-wrapper/params/zed.cfg:

# params/zed_yaml
# Parameters for Stereolabs ZED camera
---

general:
    camera_model:               'zed'

depth:
    min_depth:                  0.7             # Min: 0.3, Max: 3.0 - Default 0.7 - Note: reducing this value wil require more computational power and GPU memory
    max_depth:                  30.0            # Max: 40.0

The max_depth by default is 10 m, and I set it to 30 m, I will check how accurate is that estimation. All in all, in order to estimate further xyz objects we will need a camera with a higher baseline.

closed 2021-05-17 by Myzhar