stereolabs/zed-ros-wrapper #636: [BUG] Zed 2 sensors firmware not recognized in Ubuntu 16

carlos-argueta, 2020-10-23:
**Describe the bug**
I have both the Zed mini and Zed 2, I also have a PC with Ubuntu 18 and one with Ubuntu 16. Both cameras work fine under Ubuntu 18, but the Zed 2 fails under Ubuntu 16. After looking carefully, I noticed that under Ubuntu 16, the firmware for the sensors of the Zed 2 cannot be determined. Attached are images from both the Zed Explorer, and output from launching the zed_wrapper node. 

[image]
[image]

[image]

Notice how the firmware version cannot be determined and the ros node dies after trying to activate the IMU. I also tried running the Zed Sensor Viewer and it fails. See attached image.

[image]

**Device information:**
 - OS: Ubuntu
 - OS version: 16.04
 - CPU: i7
 - GPU Nvidia:  GTX1650 CUDA 10.2
 - ZED SDK Version: v3.2.2
 - ROS Wrapper version: v3.2.2 latest master

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the ZED_Sensor_Viewer or
2. roslaunch zed_wrapper zed2.launch
3. See error

**Expected behavior**
The zed_wrapper to run without problems

Myzhar, 2020-10-23:
Hi @carlos-argueta 
this issue is not related to the ROS Wrapper, but to the ZED SDK. Please send an email to support@stereolabs.com to open a ticket and receive direct support.

closed 2020-10-23 by Myzhar

carlos-argueta, 2020-10-24:
it makes sense, thanks a lot for the quick response.