stereolabs/zed-ros-wrapper #638: [Question] How can I retrieve the real depth instead of 32-bit encoding?

Cram3r95, 2020-10-23:
I have installed the ZED-python-api since I realized that the depth was 32-bit encoded. Nevertheless, I think it is easier. How can I obtain the distance to every object from the sensor_img topic?

Myzhar, 2020-10-23:
Hi @Cram3r95 
The "real" depth is available in the depth topic:
`depth/depth_registered`
Every pixel contains the depth in meter as floating point value.
What about the ZED-python-api?
It's not related with ROS, you must use it if you want to write your own application that retrieves data directly from a camera.

Cram3r95, 2020-10-23:
@Myzhar you are right. I thought I must combine the zed-ros-wrapper and zed-pythonapi in order to retrieve the floating point of the depth. Maybe I have not checked the right field. In the /zed/zed_node/depth/depth_registered, the field data is made up by 0-255 values, as they represent de black to White values of the images. Is this topic incorrect in order to retrieve the real value of depth?

Myzhar, 2020-10-23:
@Cram3r95 that's the correct topic.
Data in image topic are always stored in unsigned char array, you must cast them to floating point to get the value. Practically each 4 consecutive uchar values becomes a single 32bit float32 value

Cram3r95, 2020-10-24:
@Myzhar I checked that the depth rostopic is as following:

header:
  seq: 42
  stamp:
    secs: 1603377743
    nsecs: 665695708
  frame_id: "zed_left_camera_optical_frame"
height: 720
width: 1280
encoding: "32FC1"
is_bigendian: 0
step: 5120

I am searching on different forums but I am not pretty sure how to convert it to the real depth values. I observed 16UC1 (unsigned  int 1-channel). What do you recommend me?

Myzhar, 2020-10-24:
If you are writing a Python node this is a Python question, not a ROS question.

Maybe this can help you:
https://stackoverflow.com/questions/56396524/how-to-convert-a-byte-array-to-float-in-python

Like I told you before the data array is an array of unsigned char (uint8), but the encoding is float32 that means that you must group 4 consecutive 8 bit values to get a complete informative 32 bit float... each 4 8 bit values are a single 32bit depth pixel

Cram3r95, 2020-10-24:
@Myzhar I got some results after executing this test code:

"""
#!/usr/bin/env python
import rospy
import cv_bridge
from sensor_msgs.msg import Image
import numpy as np

def convert_depth_image(ros_image):
    bridge = cv_bridge.CvBridge()
    # Use cv_bridge() to convert the ROS image to OpenCV format
    try:
    #Convert the depth image using the default passthrough encoding
        depth_image = bridge.imgmsg_to_cv2(ros_image, desired_encoding="passthrough")

    except cv_bridge.CvBridgeError, e:
        print e
    #Convert the depth image to a Numpy array
    depth_array = np.array(depth_image, dtype=np.float32)
    print("real depth: ", depth_array)
    rospy.loginfo(depth_array)

def pixel2depth():
	rospy.init_node('pixel2depth',anonymous=True)
	rospy.Subscriber("/zed/zed_node/depth/depth_registered", Image,callback=convert_depth_image, queue_size=1)
	rospy.spin()

if __name__ == '__main__':
	pixel2depth()
"""

Then, I get the following results:

('real depth: ', array([[        nan,         nan,         nan, ...,  2.80438852,
         2.80404472,  2.77419257],
       [        nan,         nan,         nan, ...,  2.80395746,
         2.80408216,  2.77383542],
       [        nan,         nan,         nan, ...,  2.81517744,
         2.81418896,  2.82850385],
       ..., 
       [        nan,         nan,         nan, ...,  0.79548681,
         0.79550701,  0.80244905],
       [        nan,         nan,         nan, ...,  0.79539323,
         0.79540062,  0.8025912 ],
       [        nan,         nan,         nan, ...,  0.77282786,
         0.7728259 ,  0.77350998]], dtype=float32))

Which a priori I guess that make sense. Nevertheless, I also got an error trying to run the cv_bridge with Python3 (I need python 3 since my deep learning based object detector and multi-object tracking are python3 based). Have you ever tried to get the real depth values with no need to use the cv_bridge or any similar function or OpenCV framework?

Myzhar, 2020-10-24:
Python3 is a pain with ROS1 because it's still based on Python 2.7.
You must search online for a solution

Cram3r95, 2020-10-24:
@Myzhar you are right, thanks for your help.

Cram3r95, 2020-10-24:
@Myzhar I think that I got it, since the values are pretty similar using the cv_bridge with Python2 and Image.frombytes using PIL (in Python3).

The specific part of the code is as following:

mode_depth='F'
depth_map = Image.frombytes(mode_depth, (depth_map_rosmsg.width, depth_map_rosmsg.height), depth_map_rosmsg.data)

My mistake was that I considered each pixel value, ranged from 0 to 255, as 8bit encoding, so I set the mode_depth to L. If you check the following table:

    The mode of an image defines the type and depth of a pixel in the image. The current release supports the following standard modes:

        1 (1-bit pixels, black and white, stored with one pixel per byte)
        L (8-bit pixels, black and white)
        P (8-bit pixels, mapped to any other mode using a colour palette)
        RGB (3x8-bit pixels, true colour)
        RGBA (4x8-bit pixels, true colour with transparency mask)
        CMYK (4x8-bit pixels, colour separation)
        YCbCr (3x8-bit pixels, colour video format)
        I (32-bit signed integer pixels)
        F (32-bit floating point pixels)

It can be observed than modifying this value to F, you get the (I assume) real depth values. Again, thank you so much for your help :).

Myzhar, 2020-10-24:
You are welcome. Please close the issue if it is solved and do not hesitate to open a new one in case you need help for other problems

closed 2020-10-24 by Cram3r95