stereolabs/zed-ros-wrapper #639: [Question] Why is the odom twist covariance all 0?

tianshiz, 2020-10-23:
Hi,
 I'm using a ZED1. I noticed that the covariance for pose are filled but the twist covariance are all 0. When performing sensor fusion, should I only be trusting the pose data and not twist?

Myzhar, 2020-10-23:
Hi @tianshiz 
The twist covariance is is a null matrix because the twist info is not available

tianshiz, 2020-10-23:
ah ok, I just noticed that too. Is that something that would be available on the ZED2 since it has an IMU?

Myzhar, 2020-10-24:
It's a feature that does not depend on the presence of an IMU sensor.
To provide twist and twist covariance in `base_link` frame a frame conversion is required. 
We are studying the best way to add this feature, maybe it will be available in the future

closed 2020-11-02 by tianshiz