stereolabs/zed-ros-wrapper #642: [Question] catkin_make error

OkanoShogo0903, 2020-11-04:
- my env
  cuda 10.2, ros melodic, zedSDK 3.1, zed-ros-wrapper 3.1.x, in docker

- Executed the command
```
$ catkin_make -DCMAKE_BUILD_TYPE=Release
```

- I receive the following error
```
[ 95%] Built target zed_wrapper_node
[ 95%] Built target zed_interfaces_generate_messages
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libusb-1.0.so', needed by '/root/catkin_ws/devel/lib/libZEDNodelets.so'.  Stop.
CMakeFiles/Makefile2:2090: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
```

Myzhar, 2020-11-04:
The libusb library is missing:
`sudo apt-get install libusb-1.0-0-dev`

OkanoShogo0903, 2020-11-04:
resolved.

thank you for quick response.

closed 2020-11-04 by OkanoShogo0903