stereolabs/zed-ros-wrapper #643: [Question] roslaunch error

OkanoShogo0903, 2020-11-04:
- my env
cuda 10.2, ros melodic, zedSDK 3.1, zed-ros-wrapper 3.1.x, in docker

- Executed the command
```
$ cd /root/catkin_ws
$ rosdep install --from-paths src --ignore-src -r -y
$ catkin_make -DCMAKE_BUILD_TYPE=Release
$ source ./devel/setup.bash
$ roslaunch zed_wrapper zed2.launch
```

- I receive the following error in `roslaunch zed_wrapper zed2.launch`
```
[ERROR] [1604482019.834719668]: Failed to load nodelet [/zed2/zed_node] of type [zed_nodelets/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /root/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libturbojpeg.so.0: cannot open shared object file: No such file or directory)
[ERROR] [1604482019.834751742]: The error before refreshing the cache was: Failed to load library /root/catkin_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libturbojpeg.so.0: cannot open shared object file: No such file or directory)
```

I tried to install libturbojpeg.
I confirmed to exist the file (`/usr/local/lib/libturbojpeg.so.0`) and add `$export PATH=$PATH:/usr/local/lib` to PATH.
But i was caused same error.

Myzhar, 2020-11-04:
Hi @OkanoShogo0903 
what is your operating system?

OkanoShogo0903, 2020-11-05:
Thank you, errro was resolved!! 

I installed libturbojpeg from source, maybe it was not good.

---

> what is your operating system?
Linux

closed 2020-11-05 by OkanoShogo0903

fatemeh-mohseni-AI, 2024-01-14:
sudo apt-get install libturbojpeg-dev