stereolabs/zed-ros-wrapper #644: [Question] Using a rosbag recording of some ZED topics for offline processing

ercanburak, 2020-11-05:
Hi,

I have a rosbag file, which contains grayscale rectified images from a Zed 2 camera, namely the topics /zed2/zed_node/left/image_rect_gray and /zed2/zed_node/right/image_rect_gray. It does not contain any other Zed related topics.

Is it possible to process these grayscale rectified images offline to produce messages of other topics, such as /zed2/zed_node/pose_with_covariance or /zed2/zed_node/point_cloud/cloud_registered? If so, how can I proceed? If not, what is the minimum required set of topics that need to be recorded in rosbag beforehand to obtain such capability? 
 
Thanks,

ercanburak, 2020-11-05:
This is related to Issue #461

Myzhar, 2020-11-05:
Hi @ercanburak 
each ZED topic is not generated from other ZED topics, but it is produced processing data from the camera itself using the ZED SDK.
If you need data from a topic you must record data from that topic.
e.g. if you need `/zed2/zed_node/pose_with_covariance` then you must record it in your bag file while the ZED node is running.

An easier way to produce each topic from a recorded file is recording ZED data in SVO file using the ZED SDK recording capability. See `start_svo_recording` service:
https://www.stereolabs.com/docs/ros/zed-node/#services

ercanburak, 2020-11-05:
Hi @Myzhar , thanks for the quick reply. If I determine the required ZED topics beforehand, I will record exactly these topics when recording a rosbag file. I also record topics from other sensors in this rosbag file. If I choose to do an SVO recording for ZED instead, will I run into synchronization issues between the SVO file and topics of other sensors in the bag file?

Myzhar, 2020-11-05:
Yes, there is a big problem with using SVO, in playback we cannot simulate timestamps. It's a known limitation and we have not yet found a working solution, see https://github.com/stereolabs/zed-ros-wrapper/issues/411.

If you need data synchronized with other sensors you can only use rosbag and you must know beforehand all the topics that you will need.

ercanburak, 2020-11-05:
Okay, thank you very much for the information.

closed 2020-11-05 by ercanburak