stereolabs/zed-ros-wrapper #648: [Question] Electromagnetic interference from Zed2 with GPS antenna?

shecker, 2020-11-26:
We are witnessing some sort of interference with our Zed2 and a GPS antenna when both are in close proximity.

If the Zed2 is running and/or powered on and moved close to the GPS antenna (around 20-30cm) we are unable to get a proper GPS fix. The number of satellites immediately drops down to 0. If we move the Zed2 away we are able to pick up the satellites again (in our case around 8).

Is there any reason for this behavior and how would we prevent this from happening?
Thanks

Myzhar, 2021-01-04:
Hi, @shecker 
we faced similar issues with the ZED camera. The ZED2 has a full aluminum case that reduces the EMI distortions, but not totally. 

A user solved the interference problem by adding copper foils (not aluminum) around the case and the cable:

[image]

shecker, 2021-01-20:
Hi @Myzhar,
cool thanks for the tip. We solved by placing both further apart,

closed 2021-01-20 by shecker

flabrosse, 2021-09-24:
Having the same issue here with a ZED camera.  I put the cable in a steel braid and that helps a bit, but not completely.  I'll try copper.

For info, we have the same issue with a Intel Realsense camera and using it over USB2 rather than USB3 got rid of the noise.  Can the ZED be used over USB2?

Myzhar, 2021-09-24:
@flabrosse yes, you can use USB2, but you are limited to VGA resolution because of limited bandwidth.

PS Please do not reply to closed issues, always open a new one and add a link to the closed one.

TillBeemelmanns, 2024-03-25:
> Hi, @shecker we faced similar issues with the ZED camera. The ZED2 has a full aluminum case that reduces the EMI distortions, but not totally.
> 
> A user solved the interference problem by adding copper foils (not aluminum) around the case and the cable:
> 
> [image]

Hi @Myzhar, we are facing a similar problem.

Do you have any more details on the tests shown in the image ? 
i.e. which configuration is most effective for reducing the electromagnetic noise ?

Thanks for your help !

Myzhar, 2024-03-25:
@TillBeemelmanns the image "Test 3" shows the more reliable solution.
Please note that the new [ZED X](https://www.stereolabs.com/products/zed-x) and [ZED X One](https://www.stereolabs.com/products/zed-x-one) camera models do not suffer of this problem thanks to the GMSL2 connectivity

flabrosse, 2024-03-25:
We put the ZED camera in a metal box (aluminium, admittedly) and wrapped the USB cable in a steel braid, both end connected to ground.  That did help a bit, but not a lot.  We swapped to a ZED2i.  That is much better.  I guess the cable is better quality and the camera is fully enclosed in a metal box.  But still quite a lot of interference.

The best solution is really to move the USB cable apart from the GNSS antenna, but is not always possible (I'm having the same issue with RealSense camera on a small robot...).

Myzhar, 2024-03-25:
Yes, the USB3 communication works on a frequency that disturbs the GNSS signal.
The high rate of ZED communication amplifies this effect.
This is one of the reasons why we decided to explore other communication protocols more suitable for robotics and automotive and we created the GMSL2 ZED X family.