stereolabs/zed-ros-wrapper #651: [Question] How to export stereo images using rtabmap with ZED?

leo1200213, 2020-12-11:
Hello,

I'm trying to use **zed_rtabmap_example** to export the stereo images. (Default is left image)
And I have changed the parameter of  **_subscribe_stereo_** from _false_ to _true_ in  **/catkin_ws/src/zed-ros-examples/examples/zed_rtabmap_example/params/rtabmap.yaml** 
[image]

However, after changing the value of **_subscribe_stereo_** , a warning pop out notifying me have to let **/zed/rtabmap/** subscribe to :
   **/zed/left/image_rect,
   /zed/right/image_rect,
   /zed/left/camera_info,
   /zed/right/camera_info**  
[image]

And it seems that there is few information on the net talking about how the subscription really work in zed-ros-wrapper. So what should I do to produce stereo images in **zed_rtabmap_example**??

Thanks!

Myzhar, 2020-12-11:
Hi @leo1200213 
you must check the correct name of the topics to subscribe:
`$ rostopic list`
furthermore, control the `subscribe_stereo` parameter because you entered `ture` instead of `true`

leo1200213, 2020-12-11:
Thank for your prompt comment! Sorry for some dumb questions from the new in zed-ros-wrapper.

So if I have to subscribe to these four topics, is it necessary to write 
a publisher and subscriber just like the link suggesting? [http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29](url)

Or there is a simpler way to solve this problem?

Thanks a lot!

Myzhar, 2020-12-11:
No, you do not need to write a new node, you can simply change the topic name change in the launch files

leo1200213, 2020-12-12:
Hi @Myzhar , I have change the topics in **zed_rtabmap.launch**.
<?xml version="1.0"?>

<launch>
    <arg name="svo_file"             default="" /> 
    <arg name="stream"               default="" /> 

    <arg name="zed_node_name"        default="zed_node" />
    <arg name="camera_model"         default="zed2" />
    <arg name="publish_urdf"         default="true" />

    <arg name="camera_name"          default="zed" />

    <arg name="base_frame"           default="base_link" />

    <group ns="$(arg camera_name)">
        
        <include file="$(find zed_wrapper)/launch/include/zed_camera.launch.xml">
            <arg name="camera_name"         value="$(arg camera_name)" />
            <arg name="svo_file"            value="$(arg svo_file)" />
            <arg name="stream"              value="$(arg stream)" />
            <arg name="node_name"           value="$(arg zed_node_name)" />
            <arg name="camera_model"        value="$(arg camera_model)" />
            <arg name="base_frame"          value="$(arg base_frame)" />
            <arg name="publish_urdf"        value="$(arg publish_urdf)" />
        </include>    
  
        
        <arg name="custom_rtabmap_launch_file" default="$(find zed_rtabmap_example)/launch/include/sl_rtabmap.launch.xml"/>
        <include file="$(arg custom_rtabmap_launch_file)">
            <arg name="odom_topic"              default="$(arg zed_node_name)/odom" />
             
            <arg name="stereo_topic"               default="$(arg zed_node_name)/stereo/image_rect_color" />
            <arg name="depth_topic"             default="$(arg zed_node_name)/depth/depth_registered" />
            
            <arg name="camera_info_topic"       default="$(arg zed_node_name)/stereo/camera_info" />/
            <arg name="depth_camera_info_topic" default="$(arg zed_node_name)/depth/camera_info" />
        </include>   

        
        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find zed_rtabmap_example)/rviz/zed-rtabmap.rviz" output="screen" /> 
    </group>
</launch>

And also in **sl_rtabmap.launch.xml** .

<launch>
    
    <arg name="localization"            default="false"/>

    <arg     if="$(arg localization)" name="args" default=""/>
    <arg unless="$(arg localization)" name="args" default="--delete_db_on_start"/>

    
    <arg name="stereo_topic"               default="/zed/zed_node/stereo/image_rect_color" />
    
    <arg name="depth_topic"             default="/zed/zed_node/depth/depth_registered" />
    <arg name="camera_info_topic"       default="/zed/zed_node/stereo/camera_info" />
    
    <arg name="depth_camera_info_topic" default="/zed/zed_node/depth/camera_info" />

    <arg name="odom_topic"              default="/zed/zed_node/odom" />   
    
    

    

    
    

    
    <node name="rtabmapviz" pkg="rtabmap_ros" type="rtabmapviz" output="screen" args="" launch-prefix="">    
        
        <remap from="stereo/image"         to="$(arg stereo_topic)"/>
        <remap from="depth/image"       to="$(arg depth_topic)"/>
        <remap from="stereo/camera_info"   to="$(arg camera_info_topic)"/>

        <remap from="/left/image_rect"      to="zed/zed_node/left/image_rect_color"/>
        <remap from="/right/image_rect"     to="zed/zed_node/right/image_rect_color"/>
        <remap from="/left/camera_info"     to="zed/zed_node/left/camera_info"/>
        <remap from="/right/camera_info"    to="zed/zed_node/right/camera_info"/>

        <remap from="grid_map"          to="map" />
        <remap from="odom"              to="$(arg odom_topic)"/>
    </node> 

    
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" launch-prefix="">
        <rosparam command="load" file="$(find zed_rtabmap_example)/params/rtabmap.yaml" />
    	
        <remap from="stereo/image"         to="$(arg stereo_topic)"/>
        <remap from="depth/image"       to="$(arg depth_topic)"/>
        <remap from="stereo/camera_info"   to="$(arg camera_info_topic)"/>

        <remap from="/left/image_rect"      to="zed/zed_node/left/image_rect_color"/>
        <remap from="/right/image_rect"     to="zed/zed_node/right/image_rect_color"/>
        <remap from="/left/camera_info"     to="zed/zed_node/left/camera_info"/>
        <remap from="/right/camera_info"    to="zed/zed_node/right/camera_info"/>

        <remap from="grid_map"          to="map" />
        <remap from="odom"              to="$(arg odom_topic)"/>

        
        <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
        <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
        <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>

    

</launch>

I have taken **"/left/image_rect"** , **"/right/image_rect"** , **"/left/camera_info"** and **"/right/camera_info"** into account in the **sl_rtabmap.launch.xml**  file.,  but the warning of subscription still exists like before. 

I'm not sure these changes I made are right or not.

Myzhar, 2020-12-12:
> I have taken "/left/image_rect" , "/right/image_rect" , "/left/camera_info" and "/right/camera_info" into account in the sl_rtabmap.launch.xml file., but the warning of subscription still exists like before.

Yes, because the correct topic is `/left/image_rect_color` and `/right/image_rect_color`.

Here you can find the documentation with the explanation of each topic:
https://www.stereolabs.com/docs/ros/zed-node/#published-topics

When you have this kind of problems, you can try to use the command `$ roswtf` to understand what's happening

closed 2021-05-17 by Myzhar