stereolabs/zed-ros-wrapper #652: Can we limit publishing streams?

Russ76, 2020-12-18:
By default, this wrapper publishes many data streams. Doesn't this burden the CPU to publish more data than is required? Can we control the streams or turn some off?

Myzhar, 2020-12-18:
Hi @Russ76 
the wrapper controls the subscribers to each stream and enables the publishing of the relative topic only if there is a subscriber for it. In this way no data is published if it's not required.

Russ76, 2020-12-18:
Maybe this has been changed lately? For a couple years ago, I ran this command, and it produced a list of 84 topics from camera. Only 5 had subscribers.

Russ

nvidia@tegra-ubuntu:~$ rostopic list -v

Published topics:
 * /camera/rgb/image_rect_color/theora/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /camera/rgb/image_raw_color/compressedDepth [sensor_msgs/CompressedImage] 1 publisher
 * /camera/left/image_rect_color/compressed [sensor_msgs/CompressedImage] 1 publisher

Myzhar, 2020-12-18:
The behavior of the ZED ROS Wrapper is the same from since I'm using it (even before I started working on it):
1) all the available topics are ADVERTISED 
2) only the subscribed topics are PUBLISHED
Advertising a topic is required, otherwise the subscribers could not know what to subscribe.
Advertising a topic does not mean that the messages are published.
You can analize the code, you will notice that a message is published on the relative topics only if the number of subscribers is major than 0.
In the same way, all the elaborations required to create the message before publishing are performed only if the topic is subscribed, unless they are not required by other parts of the wrapper (for example, the odometry is calculated even if the odom topic is not subscribed because it's required by the TF).
To conclude, the `rostopic list` command displays the list of all the advertised topics, not the list of all the published messages on those topics.

Russ76, 2020-12-20:
OK, good to know! Then it wasn't burdening the system like I suspected. I think next time I'll try my robot with two Jetson TX1s, one connected to ZED, and the other controlling robot movement and running the SLAM. I like the Xavier but don't have one, while I have two TX1s!

closed 2021-05-17 by Myzhar