stereolabs/zed-ros-wrapper #653: [Question]

zoombinis, 2020-12-20:
```shell
roslaunch zed_nodelet_example zed_sync_nodelet.launch camera_model:=zed sub_imu:=false sub_mag:=false sensing_mode:=1 quality:=4 depth_stabilization:=1
```

How can I set the depth parameters from the command line so that I do not need to tweak any launchfiles?

Thanks!

zoombinis, 2020-12-21:
Here's a work around:

```shell
      sed -i s/max_depth/#max_depth/ $ROS_WS/src/zed-ros-wrapper/zed_wrapper/params/zed.yaml && \
      echo '    max_depth: 40.0' >> $ROS_WS/src/zed-ros-wrapper/zed_wrapper/params/zed.yaml && \
      echo '    quality: 4' >> $ROS_WS/src/zed-ros-wrapper/zed_wrapper/params/zed.yaml && \
      echo '    sensing_mode: 1' >> $ROS_WS/src/zed-ros-wrapper/zed_wrapper/params/zed.yaml && \
      roslaunch zed_nodelet_example zed_sync_nodelet.launch camera_model:=zed sub_imu:=false sub_mag:=false
```

Myzhar, 2020-12-21:
Hi @zoombinis 
Have you tried simply adding the relative PREFIX to each parameter that you want to change?

`roslaunch zed_nodelet_example zed_sync_nodelet.launch camera_model:=zed sub_imu:=false sub_mag:=false depth.sensing_mode:=1 depth.quality:=4 depth.depth_stabilization:=1`

PS for a robotic application setting sensing_mode to 1 is not suggested. That is a mode created for VR and AR applications.

zoombinis, 2020-12-21:
yes I tried that and it didnt work. Why isn't that mode recommended? I'm trying to get rid of all the NAN INF values, as otherwise I can't post-process depth information

closed 2021-05-17 by Myzhar