stereolabs/zed-ros-wrapper #654: How to replace NaN and -Inf, Inf depth values?

zoombinis, 2020-12-21:
I have a use-case where I need to rely on opencv C++ post-processing methods like min/Max Locator and histogram calculations, which cannot run correctly with NaN or -Inf, Inf values in the dataset.

FILL mode is supposed to not have invalid values:

- https://github.com/stereolabs/zed-ros-wrapper/issues/573#issuecomment-648086176
- https://github.com/stereolabs/zed-ros-wrapper/issues/67#issuecomment-247636012

But when I run with fill mode and Ultra quality I still get them. So what is the recommended way to filter inf, -inf and nan to be able to run post processing on the data like `cv::calcHist` and get meaningful results? Luckily there's `cv::patchNaN` but still there's nothing to handle -inf and inf. Is slowly iterating through each pixel of the mat my only option or is there a "mode" for the zed that can publish zeros instead of NaN/Inf?:

```cpp
cv::Mat DepthFusion::patch_invalid(cv::Mat& depth, float replacement_value)
    {
        cv::Mat patched = depth.clone();
        for(int r = 0; r < patched.rows; r++)
        {
            auto* row = patched.ptr<float>(r);
            for(int c = 0; c < patched.cols; c++) {
                if (cvIsInf(row[c]) || cvIsNaN(row[c])) {
                    row[c] = replacement_value;
                }
            }
        }

        return patched;
    }
```

Myzhar, 2020-12-21:
Inf and Nan value have a meaning in the depth map.
FILL mode is not the solution to remove them.
FILL mode is used to fill the holes, but it's not good in robotics because it adds "not real" measures.
The correct way of working is processing the depth map to extract min and max ignoring inf and nan values, instead of "patching" the depth map.

zoombinis, 2020-12-21:
@Myzhar I agree but opencv [has a bug](https://stackoverflow.com/a/41759433) with support for masking inf, -inf, or NaN (suggested fix is to replace as I am doing).

So the only problem is iterating through replacing those values with 0, which I was hoping there was a switch in the zed to do for me.

closed 2021-05-17 by Myzhar