stereolabs/zed-ros-wrapper #655: Catkin_make Error

ManChrys, 2020-12-29:
Hello guys,

First of all Merry Christmas and i wish health and happiness to all!!!!

So when im catkin_make i have this error :

[image]

nvcc --version

[image]

So what's my fault ?

Thanks in advance.

Myzhar, 2020-12-29:
Hi @ManChrys 
you installed CUDA 11.1 on your system, but the ZED SDK that you are using probably is for CUDA 11.0.
Is it possible that you updated CUDA after installing the ZED SDK for a previous version?
Maybe that you installed the correct ZED SDK, but you previously compiled the ZED Wrapper using a ZED SDK for CUDA 11.0.

In every case the solution is simple: 
1. download the latest ZED SDK for CUDA 11.1
2. clean the cache of the catkin workspace deleting the `build` and the `devel` folders
3. recompile.

ManChrys, 2020-12-30:
Fast response and 100% helpful answer again!!!

My cuda's version was 11.1 my Zed sdk version was ZED_SDK_Ubuntu18_cuda11.1_v3.3.3.run but when i installed Zed it asked me if i wanted to install cuda 11.1. My fault was probably that i accepted the install! I followed your 3 steps solution and it worked fine again.

Thank you a lot!

closed 2020-12-30 by ManChrys