stereolabs/zed-ros-wrapper #657: [Question]  OpenMP not found

Hagaik92, 2021-01-02:
Hello,
im having a problem while using catkin_make . The error i get is the one described above:
Cmake Error at zed-ros-wrapper/zed_nodelets/CmakeLists.txt:20
OpenMp not found please install it to improve performance : 'sudo apt install libomp-dev'

when running this command above, it says that the newest version is already installed.

im running the program on jetson Xavier NX . so if anyone has similar kit and working configuraion that could help as well.
Many Thanks

Myzhar, 2021-01-04:
Hi @Hagaik92 
can you please copy and paste the complete output of `catkin_make` command?

Hagaik92, 2021-01-04:
Hello,
Thanks for your email.
i have removed and installed jetpack 4.4.1 and it fixed it.
however, i still dont manage to use Zed2 on my Xavier NX with Ros . do you
have by any chance a working configuration for using Zed2 to detect objects
on Xavier NX and publish the output to Ros?
Thanks

Myzhar, 2021-01-04:
Many users can use the ZED2 with ROS on the Xavier NX without any kind of issue.
Have you followed correctly the [installation procedure](https://github.com/stereolabs/zed-ros-wrapper#getting-started)? 
For object detection you can read here:
https://www.stereolabs.com/docs/ros/object-detection/

Hagaik92, 2021-01-04:
Yes i did. followed the procedure and installed the zed-ros-wrapper.
However, i want to use custom object detection with a trained Yolov3 or
yolov4 model- and I'm having problems doing that.
do you have a good suggestion doing that?
Thanks

closed 2021-05-17 by Myzhar