stereolabs/zed-ros-wrapper #658: [Question] Is ros noetic support coming soon?

tianshiz, 2021-01-04:
Hi,
  Since cuda11 is now available to 20.04 ubuntu, does stereolabs have an approximate date for when the zed-wrapper will support ros noetic?
  I noticed that there is a zed-docker file for ros-devel already, does that mean the wrapper already works?
https://github.com/stereolabs/zed-docker/blob/master/3.X/ubuntu/ros-devel/Dockerfile

Myzhar, 2021-01-04:
We have not yet tested the ROS wrapper with ROS Noetic.
We did not because ROS Noetic is available precompiled only for Ubuntu 20.04 and the Nvidia Jetson boards, the only embedded boards with CUDA capabilities, are still based on Ubuntu 18.04.
However the wrapper is open source and you can try to compile it for ROS Noetic, eventually fixing what does not work, that I suppose will be really few.

Myzhar, 2021-05-17:
A development branch for ROS Noetic is available here:
https://github.com/stereolabs/zed-ros-wrapper/tree/dev_noetic

closed 2021-05-17 by Myzhar