stereolabs/zed-ros-wrapper #660: [BUG] Topics are not publishing while running Zed2 with object detection on Jetson Xavier.

payamn, 2021-01-11:
I am running multiple Zed2 mounted on a Jetson Xavier. The problem is if I run everything locally (on Jetson Xavier) it works fine but as soon as running them over the network (Set Rosmaster to another system connected over LAN) it will not publish objects unless I subscribe to /point_cloud/cloud_registered topic on each camera. 

ROS_MASTER_URI, ROS_HOSTNAME and ROS_IP set correctly in both Xavier and NUC. As when I echo topics they will show up inside the NUC.

**Device information:**
 - OS: Ubunutu
 - OS version: [e.g. 18.04]
 - CPU: [Jetson Xavier and Nuc]
 - ZED SDK Version: [e.g. v3.3]
 - ROS Wrapper version: [latest master]
 - 
When not subscribing to the pointcloud topic:
[image]

When subscribing to the pointcloud topic:
[image]

Myzhar, 2021-01-11:
Hi @payamn 
thank you for reporting this problem.
Have you verified if this happens only with the Jetson Xavier?

fafux, 2021-01-13:
I have the same problem running locally and not with a Jetson Xavier. If I launch the zed2.launch with the object detection enabled in the config file, the object detector is not publishing anything until I subscribe to the point cloud.

Myzhar, 2021-01-13:
Hi @fafux 
can you please verify if this happens only on the Xavier?

fafux, 2021-01-13:
I run the code on a NVIDIA Corporation GP104M GeForce GTX 1080 and I faced the same problem

udomi, 2021-01-27:
I am having the same problem. Using the Jetson Nano (4.4) with ZED2 (3.3). When running roslaunch zed_wrapped zed2.launch (object detection enabled) I get no objects on /zed2/zed_node/obj-det/objects topic. Only when I open RVIZ and add odom and ../object_markers topic the /zed2/zed_node/obj-det/objects shows informations on detected objects. 
Any ideas?

Myzhar, 2021-01-27:
@udomi the issue will be fixed soon. It's scheduled.

hagonzalezd, 2021-01-28:
Hello @Myzhar, I'm also having the same issue in the Jetson Xavier (4.4), and ZED2 with SDK 3.4. The obj-det topics only publish data when RVIZ is used. These topics are empty when launching the zed wrapper / zed2.launch directly, which is strange because the display_rviz launch file also calls the same wrapper. I highly appreciate your help with this issue.

Myzhar, 2021-01-28:
I was able to replicate the issue and I can add more information.
The `obj_det/objects` topic is currently published only if at least one of the following topics is subscribed:
* confidence_map
* depth and its camera_info
* disparity
* odom
* pointcloud
* pose

The bug is under investigation to understand the cause and how to fix it. Stay tuned for updates

hagonzalezd, 2021-01-28:
Hello @Myzhar thanks for your prompt response. This is already a good workaround to get the objects until the problem is solved.

Thanks

udomi, 2021-01-28:
Hello Walter,

thanks for the update.
Best Regards

Udo

Myzhar, 2021-01-29:
@udomi @hagonzalezd @payamn @fafux can I ask you to test the branch `devel_3.4` to confirm that the bug is fixed?
https://github.com/stereolabs/zed-ros-wrapper/tree/devel_v3.4

udomi, 2021-01-30:
Hello Walter,

I reinstalled zed_wrapper with git clone
https://github.com/stereolabs/zed-ros-wrapper.git
This seems to be still the buggy version. How can I download the latest
version in devel_v3.4?

Thanks

Udo

Myzhar, 2021-01-30:
@udomi before compiling you must switch to the `devel_3.4` branch. It was not required to reinstall everything. This is the procedure:

```
$ roscd zed_wrapper 
$ git pull
$ git checkout devel_v3.4
$ cd ..
$ cd ..
$ catkin_make
```

With `roscd` you enter directly in the folder of the package, with `pull` you update your local repo with the latest remote modifications, with `checkout` you change the current branch (from `master` to `devel_v3.4`)

udomi, 2021-01-30:
Hello Walter,

thanks for the instructions.
First test with devel_v3.4 looks good.
rostopic /zed2/zed_node/obj_det/objects  is working now :-)

One more question:
What is the dimension of the position x,y,z value in
the /zed2/zed_node/obj_det/objects topic?
I was expecting meter but it seems decimeter?

label: "Vehicle"
    label_id: 1
    confidence: 61.6971359253
    position:
      x: 2.46579647064
      y: -0.125977739692
      z: -0.157986491919
    linear_vel:
      x: 0.0997988805175
      y: -0.00473020598292
      z: -0.00112907297444
    tracking_state: 1

fafux, 2021-02-23:
@Myzhar the devel_v3.4 branch is working properly. Thanks

Myzhar, 2021-02-23:
@fafux thank you for the feedback

Myzhar, 2021-02-25:
The issue has been fixed and merged in the master branch with release v3.4.x

closed 2021-02-25 by Myzhar

Baldins, 2022-09-29:
Hello, 

I have the same issue. Everything is up to date but /zed/nodelet/obj_det/objects is not publishing any objects. 
Any help on this?

udomi, 2022-09-30:
Hello,

as far as I remember a Software Update did fix the Problem.

Best regards

Udo

Myzhar, 2022-09-30:
@Baldins this is a closed issue.
Please create a new issue providing all the required information.