stereolabs/zed-ros-wrapper #662: [Question] ZED2 Attempt memory reboot

Cyuan-Ming, 2021-01-15:
I can use it well in few hours ago.
but. now i got a problem about this. Can anyone tell me how to fix it?

[image]

Myzhar, 2021-01-15:
Hi @Cyuan-Ming 
can you provide more information about what node you are running and about your configuration?

Cyuan-Ming, 2021-01-15:
Hi @Myzhar 
Thanks for your quick reply

I use this command roslaunch zed_wrapper zed2.launch

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: False
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path: 
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed2/zed_node/pos_tracking/floor_alignment: False
 * /zed2/zed_node/pos_tracking/imu_fusion: True
 * /zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed2/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed2/zed_node/pos_tracking/map_frame: map
 * /zed2/zed_node/pos_tracking/odometry_frame: odom
 * /zed2/zed_node/pos_tracking/path_max_count: -1
 * /zed2/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed2/zed_node/pos_tracking/publish_map_tf: True
 * /zed2/zed_node/pos_tracking/publish_tf: True
 * /zed2/zed_node/pos_tracking/two_d_mode: False
 * /zed2/zed_node/pub_frame_rate: 15.0
 * /zed2/zed_node/saturation: 4
 * /zed2/zed_node/sensors/publish_imu_tf: True
 * /zed2/zed_node/sensors/sensors_timestamp_sync: False
 * /zed2/zed_node/sharpness: 4
[... 62 lines cut ...]
[ INFO] [1610724913.747541836]:  * confidence_frame		-> zed2_left_camera_frame
[ INFO] [1610724913.747595488]:  * confidence_optical_frame	-> zed2_left_camera_optical_frame
[ INFO] [1610724913.749163846]:  * Broadcast odometry TF	-> ENABLED
[ INFO] [1610724913.750529956]:  * Broadcast map pose TF	-> ENABLED
[ INFO] [1610724913.751832264]:  * Broadcast IMU pose TF	-> ENABLED
[ INFO] [1610724913.752802123]:  * [DYN] Depth confidence	-> 50
[ INFO] [1610724913.753679231]:  * [DYN] Depth texture conf.	-> 100
[ INFO] [1610724913.754567235]:  * [DYN] pub_frame_rate		-> 15 Hz
[ INFO] [1610724913.755193115]:  * [DYN] point_cloud_freq	-> 15 Hz
[ INFO] [1610724913.755809839]:  * [DYN] brightness		-> 4
[ INFO] [1610724913.756426727]:  * [DYN] contrast		-> 4
[ INFO] [1610724913.757016310]:  * [DYN] hue			-> 0
[ INFO] [1610724913.757723467]:  * [DYN] saturation		-> 4
[ INFO] [1610724913.758353198]:  * [DYN] sharpness		-> 4
[ INFO] [1610724913.758915239]:  * [DYN] gamma			-> 8
[ INFO] [1610724913.759469043]:  * [DYN] auto_exposure_gain	-> ENABLED
[ INFO] [1610724913.760621229]:  * [DYN] auto_whitebalance	-> ENABLED
[ INFO] [1610724913.766904894]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1610724913.767145618]:  *** Opening ZED 2...
[ZED][Sensors] Attempt memory reboot
[ INFO] [1610724915.418278689]: ZED connection -> CAMERA NOT DETECTED
[ZED][Sensors] Attempt memory reboot
[ INFO] [1610724918.912179091]: ZED connection -> CAMERA NOT DETECTED

I have try to run ZED_Depth_Viewer etc. but they also can't work.

Myzhar, 2021-01-15:
Have you tried rebooting the system and then opening `ZED_Explorer`?
If it does not work, please run `ZED_Diagnostic` and then send the result report to support@stereolabs.com to open a support ticket.

Cyuan-Ming, 2021-01-15:
> Have you tried rebooting the system and then opening `ZED_Explorer`?
> If it does not work, please run `ZED_Diagnostic` and then send the result report to [support@stereolabs.com](mailto:support@stereolabs.com) to open a support ticket.

Yes, i have tried. The problem still exist. 
Maybe it my device problem. Thanks a lot.

closed 2021-01-15 by Cyuan-Ming