stereolabs/zed-ros-wrapper #663: [BUG] symbol lookup error: undefined symbol TransformBroadcaster

Garrus007, 2021-01-15:
**Describe the bug**
I downloaded and built v.3.2.x release. When I run zed_wrapper I got the following issue:
```
roslaunch zed_wrapper zed.launch
[ INFO] [1610745853.494404690]: Initializing nodelet with 6 worker threads.
/home/nvidia/ros/devel/lib/zed_wrapper/zed_wrapper_node: symbol lookup error: /home/nvidia/ros/devel/lib//libZEDNodelets.so: 
undefined symbol: _ZN7tf2_ros20TransformBroadcasterC1Ev
================================================================================
REQUIRED process [zed/zed_node-3] has died!
process has died [pid 21007, exit code 127, cmd /home/nvidia/ros/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/nvidia/.ros/log/9ca0a3fc-5779-11eb-8a35-00044b8cfe9f/zed-zed_node-3.log].
log file: /home/nvidia/.ros/log/9ca0a3fc-5779-11eb-8a35-00044b8cfe9f/zed-zed_node-3*.log
Initiating shutdown!
================================================================================

```

But when I download and build latest master, zed_wrapper works correctly. 

**Device information:**
 - OS: Ubuntu
 - OS version:  18.04 (JetPack 4.4.1)
 - CPU: Jetson TX2
 - GPU Nvidia: Jetson TX2
 - ZED SDK Version: 3.3.3
 - ROS Wrapper version: v.3.2.x
 - ROS version: melodic

**To Reproduce**
Steps to reproduce the behavior:
1. Install JetPack 4.4.1, ZED SDK 3.3.3, ROS Melodic 
2. Download ZED Wrapper v.3.2.x, unpack source to the <catkin_workspace>/src
3. Install dependencies using rosdep and build ZED Wrapper
```bash
cd <catkin_workspace>
rosdep install --from-paths src --ignore-src  -y
catkin_make
```
4. Launch ZED wrapper
```bash
roslaunch zed_wrapper zed.launch
```

**Expected behavior**
ZED Wrappers starts correctly
```
[ INFO] [1610747033.097316248]: Initializing nodelet with 6 worker threads.
nvbuf_utils: Could not get EGL display connection
[ INFO] [1610747033.187430666]: ********** Starting nodelet '/zed/zed_node' **********
[ INFO] [1610747033.187591753]: SDK version : 3.3.3
[ INFO] [1610747033.187715016]: *** PARAMETERS ***
[ INFO] [1610747033.188556901]:  * Camera Name			-> zed
...
[ INFO] [1610747033.253445862]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1610747033.253615621]:  *** Opening ZED...
[ INFO] [1610747033.940128938]: ZED connection -> CAMERA NOT DETECTED
```
Note: in my case I don't connect ZED, it doesn't matter to reproduce this issue

Myzhar, 2021-01-16:
Hi @Garrus007 
How did you install ROS? 
It seems that you are missing the packages handling TF broadcasting, usually it's automatically installed with the default installation

Garrus007, 2021-01-16:
This symbol is from TF library. [See](http://docs.ros.org/en/kinetic/api/tf/html/c++/classtf_1_1TransformBroadcaster.html). I installed `ros-melodic-ros-base` and then installed missing packages. I make sure that r`os-melodic-tf`, `ros-melodic-tf2` and `ros-melodic-tf2-ros` are installed. 

Moreover,
1. I found the symbol _ZN7tf2_ros20TransformBroadcasterC1Ev in one of the TF shared libraries
2. When I cloned the zed-wrapper master, instead of release and built it, the problem is gone

How I checked for symbols:
```bash
/opt/ros/melodic/lib$ for x in $(find /opt/ros/melodic/lib/ -name "*.so"); do  nm -D $x | grep _ZN7tf2_ros20TransformBroadcasterC1Ev && echo $x; done

                 U _ZN7tf2_ros20TransformBroadcasterC1Ev
/opt/ros/melodic/lib/libros_filter.so
                 U _ZN7tf2_ros20TransformBroadcasterC1Ev
/opt/ros/melodic/lib/libtf.so
0000000000086928 T _ZN7tf2_ros20TransformBroadcasterC1Ev
/opt/ros/melodic/lib/libtf2_ros.so
                 U _ZN7tf2_ros20TransformBroadcasterC1Ev
/opt/ros/melodic/lib/librobot_state_publisher_solver.so
```
So you can see that the symbol in the `libtf2_ros.so`

closed 2021-05-17 by Myzhar