stereolabs/zed-ros-wrapper #665: [BUG]

shecker, 2021-01-20:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04 
 - CPU: Jeston Xavier AGX
 - GPU Nvidia:  Tegra 
 - ZED SDK Version: 3.3.1
 - ROS Wrapper version: 3.2.x
  
**To Reproduce**
Steps to reproduce the behavior:
When running the ZED with the ros_wrapper in a high dynamic environment it crashes with the following error:

```
 [INFO] [1611068185.679496640]: Published static transform 'zed2_imu_link' -> 'zed2_left_camera_frame'
terminate called after throwing an instance of 'Bayesian_filter::Numeric_exception'
  what():  S not PD in observe
================================================================================REQUIRED process [zed2/zed_node-3] has died!
process has died [pid 8243, exit code -6, cmd /mnt/Xavier1TB/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/aegis/.ros/log/77074778-5a66-11eb-86cc-01e97051a9ca/zed2-zed_node-3.log].
log file: /home/aegis/.ros/log/77074778-5a66-11eb-86cc-01e97051a9ca/zed2-zed_node-3*.log
Initiating shutdown!
================================================================================
[zed2/zed_node-3] killing on exit
[zed2/zed2_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

**Expected behavior**
It should not crash... If we run the ZED SDK samples in the same highly dynamic environment (for example birds eye view object detection) then we do not see the ZED crash and the sample works with no problem. Only when using the ros_wrapper.
Also we do not observe this behaviour when using the ros_wrapper in a static environemnt.
It is only observed when using the ros_wrapper in a highly dynamic environemnt (automotive application).

It would really be great if this issue could be fixed quickly!

Thanks a lot!

Myzhar, 2021-01-20:
Hi @shecker 
What do you mean with highly dynamic environment? Can you provide a screenahoot as example?
What module of the wrapper have you enabled?
Does the crash happens suddenly as soon as the wrapper starts of after a while?

shecker, 2021-01-20:
Hi Myzhar

> What do you mean with highly dynamic environment? 
It means that the Zed is mounted on a motorcycle and we drive around. 

> What module of the wrapper have you enabled?
see the added common.yaml and zed2.yaml below

Does the crash happens suddenly as soon as the wrapper starts of after a while?
No it happens after a while, say between 10s and 1 minute. 

common.yaml
```
# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             20.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           1.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            30                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false

video:
    img_downsample_factor:      1.0                             # Resample factor for images [0.01,1.0] The SDK works with native image sizes, but publishes rescaled image.
    extrinsic_in_camera_frame:  true                            # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]

depth:
    quality:                    1                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

pos_tracking:
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'
    odometry_frame:             'odom'
    area_memory_db_path:        ''
    area_memory:                false                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true

mapping:
[... 4 lines cut ...]
```

zed2.yaml
```
general:
    camera_model:               'zed2'

depth:
    min_depth:                  1.5            # Min: 0.2, Max: 3.0 - Default 0.7 - Note: reducing this value wil require more computational power and GPU memory
[... cut ...]
    confidence_threshold:       50              # Minimum value of the detection confidence of an object [0,100]
    object_tracking_enabled:    true            # Enable/disable the tracking of the detected objects
    people_detection:           true            # Enable/disable the detection of persons
    vehicle_detection:          true           # Enable/disable the detection of vehicle                                                                                                              
```

Myzhar, 2021-01-20:
It seems that the crash happens inside the SDK. Can I ask you to monitor the RAM usage checking if it reaches the limit before the crash?

shecker, 2021-01-20:
Hi @Myzhar 
RAM seems to be fine, it's just the CPU that is taking a lot of load. See sample ps output for start, middle, last one before crash.

     PID         DATE      TIME      ELAPSED    CPU    MEM       RSS     VSIZE       CMD
    8461   2021-01-20  16:14:53        00:10   79.8    3.3   1101384  36973768  /mnt/Xavier1TB/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/aegis/.ros/log/3798dac6-5b32-11eb-80fe-e33071a21105/zed2-zed_node-3.log
    8461   2021-01-20  16:16:22        01:40    121    7.1   2334944  38619476  /mnt/Xavier1TB/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/aegis/.ros/log/3798dac6-5b32-11eb-80fe-e33071a21105/zed2-zed_node-3.log    
    8461   2021-01-20  16:17:54        03:11    121    7.1   2336384  38619476  /mnt/Xavier1TB/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/home/aegis/.ros/log/3798dac6-5b32-11eb-80fe-e33071a21105/zed2-zed_node-3.log

Could there be some issue regarding  pose estimation or mapping? I notived that when I start the node while riding fast it usually crashes a lot faster (within seconds) vs. when launching node and waiting stationary with the bike for aroun 20s and then starting to drive. 
Thanks

Myzhar, 2021-01-20:
The log is automatically added by ROS, but it's not created/filled if not explicitly enabled in the launch file.

It could be really important for us to understand the issue if you can record an SVO replicating the crash configuration and you share it with us (you can open a ticket by writing an email to support@stereolabs.com).

shecker, 2021-01-20:
Okay, just so I understand correctly, I will record an SVO file using the SDK, no ros, and then send that to you with my params and yaml configuration?

Myzhar, 2021-01-20:
Yes, that's perfect :+1: 
Thank you
PS please share the files using an external sharing service (Dropbox, Google Drive,...)

shecker, 2021-01-20:
Quick update, with a recorded SVO there is no crashing... 
I'm currently uploading the SVO to the cloud and will share but I don't think we will be able to debug with that as the error does not show. I'm running the SVO on the Jetson with the exact parameters and the same other nodes that usually run. 
What is the difference between an SVO and live recording for the ros_wrapper?

Myzhar, 2021-01-21:
@shecker please update me as soon as you test something different about power supplying to understand if that solves the problem or not

shecker, 2021-01-21:
@Myzhar power supply did not change anything, it still crashes. 
I wrote you an email.

closed 2021-05-17 by Myzhar

jihirshuN, 2022-07-07:
Hi. We are facing the same issue. For us, it crashes when we approach the camera too fast and up close. Did it get resolved ? If yes, can you please let us know the solution ? Thanks.

VibAltekar, 2023-09-27:
@Myzhar @shecker @jihirshuN do you find any resolution? Still seeing this problem now

Myzhar, 2023-09-27:
@VibAltekar What SDK version? 
Please do not reply to closed issues.
Create a new one, add all the required information and eventually tag other issues as a reference.

jihirshuN, 2023-10-02:
> @Myzhar @shecker @jihirshuN do you find any resolution? Still seeing this problem now

Decreasing the frame rate helped.