stereolabs/zed-ros-wrapper #666: [Question] I can't set the resolution and frame rate of zed camera

heshamkhaledd, 2021-01-23:
I am making a ROS project that uses the Zed Camera and I want to adjust its parameters (Resolution and Frame Rate).
It was stated in the docs that to change those parameters I should change params/common.yaml file.

I already did that, but the it doesn't reflect on the camera when I launch it. It kept running at VGA@15 FPS no matter how I change those parameters.

I launched the launch file using `roslaunch zed_wrapper zed.launch`

Can you help me with that? I want to run it on HD720 @ 60 FPS

Kindly, check my question for further screenshots of the issue on stack overflow.
https://stackoverflow.com/questions/65854527/zed-wrapper-quality-and-frame-rate

Myzhar, 2021-01-23:
Hi @heshamkhaledd 
Can you please test if you can change resolution using `ZED_Explorer`?
Try also to run `ZED_Diagnostic` and copy/paste here the report it generates

heshamkhaledd, 2021-01-24:
Yes, I tested the `Zed_Explorer` and I could change to the resolutions I want (HD720@60FPS) and ever higher

For the `Zed_Diagnostic` here is the report

{
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 0,
        "firmware": 1523,
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@60": {
                "dirty lenses": 45,
                "dropCount": 5,
                "fps (input)": 60,
                "fps (output)": 60,
                "height (input)": 720,
                "height (output)": 720,
                "imageCount": 495,
                "initCode": true,
                "initialization": "Ok",
[... 105 lines cut ...]
                "iSerial": 0,
                "idProduct": "0xf582",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/3",
                "idProduct": "0xb728",
                "idVendor": "0x0bda"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 62850,
        "ZEDDetected": true,
        "valid": [
            "<b>USB Bandwidth: </b> OK"
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V10.0.326",
        "DiagnosticZEDSDKVersion": "3.0.0",
        "InstalledZEDSDKVersion": "Unknown",
        "initResult": 0,
        "loaded": true
    }
}

Myzhar, 2021-01-24:
This is indeed really weird, ZED_Explorer is OK and the diagnostic report does not reflect any kind of problem.
Have you, for some reason, overwritten the parameters in the launch file?

matlabbe, 2021-03-18:
In common.yaml, there is also a `pub_frame_rate` parameter (default 15) to limit the publishing rate even if the `general/grab_frame_rate` is > 15:
https://github.com/stereolabs/zed-ros-wrapper/blob/3b3a075806e18f169e8b1846d4a4ea9bcd7c6c31/zed_wrapper/params/common.yaml#L19-L27
I cannot reproduce the `resolution` issue though, it seems to work with my camera (tested on zedm).

closed 2021-05-17 by Myzhar