stereolabs/zed-ros-wrapper #668: [Question]

cnpcshangbo, 2021-02-09:
Hello,
When I try to run ```catkin_make -DCMAKE_BUILD_TYPE=Release -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs       -DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"```, I encountered this error message:
```
CMake Error at /usr/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:218 (message):
  Could NOT find CUDA (missing: CUDA_CUDART_LIBRARY) (found suitable version
  "10.0", minimum required is "10")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:582 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.19/Modules/FindCUDA.cmake:1121 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:72 (find_package)
  zed-ros-wrapper/zed_nodelets/CMakeLists.txt:25 (find_package)

-- Configuring incomplete, errors occurred!
See also "/root/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/root/catkin_ws/build/CMakeFiles/CMakeError.log".
```
It seems CUDA 10.0 and 10 means the same. Do you have some suggestions?
Thanks.

Myzhar, 2021-02-09:
@cnpcshangbo Try adding -DCUDA_CUDART_LIBRARY=/usr/local/cuda/lib64/libcudart.so when you run catkin_make command.

cnpcshangbo, 2021-02-11:
Thanks. It works.

closed 2021-02-11 by Myzhar