stereolabs/zed-ros-wrapper #670: [BUG] Duplicate timestamps on imu_raw

precyon, 2021-02-11:
**Describe the bug**
A clear and concise description of what the bug is.

I am mostly interested in the raw imu and image data from the ZED. 

I see duplicate timestamps in the `imu/data_raw` topic. The following plot from PlotJuggler helps visualize this (the software let's you compute and plot differences between fields in the topic). Notice the frequent jumps to 0. This trend continues throughout the rest of the bag.
[image]

And this, of course, is verified by visually inspecting the timestamp field. 
[image]

The bag file, for your inspection, is [here](https://drive.google.com/drive/folders/119zGqI7wnyliqDPntLek6V8uM_dJG49i). 

Also as I understand, the IMU publish rate should be 400 Hz as defined by the `mSensPubRate` variable but I see the actual publish rate to be close 600Hz for raw data (and 550Hz for filtered IMU data), some of which are duplicates. I am therefore struggling to understand the actual publish rate of the unique IMU values.

I did look at #577 which is closed. I am using the latest master which has these changes and still see this particular issue.

**Device information:**
 - OS: [e.g. Ubuntu]: Ubuntu
 - OS version: [e.g. 18.04]: 18.04
 - CPU: [e.g. i7, Jeston Nano, Jetson TX2]: i7
 - GPU Nvidia:  [e.g. GTX1080, RTX1070]: GTX 1050
 - ZED SDK Version: [e.g. v3.1.2, v2.8.5]: v3.4
 - ROS Wrapper version: [e.g. v2.8.5, latest master] latest master

**To Reproduce**
Nothing special. Just run the `zedm.launch`

**Expected behavior**
A clear and concise description of what you expected to happen.

Unique timestamps

**Screenshots**
If applicable, add screenshots to help explain your problem.
(already added above)

**Additional context**
1. `common.yaml`
```
# params/common_yaml
# Common parameters to Stereolabs ZED and ZED mini cameras
---

# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           50                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             15.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           15.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            30                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false

video:
    img_downsample_factor:      1.0                             # Resample factor for images [0.01,1.0] The SDK works with native image sizes, but publishes rescaled image.
    extrinsic_in_camera_frame:  true                            # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]

depth:
    quality:                    1                               # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA
    sensing_mode:               0                               # '0': STANDARD, '1': FILL (not use FILL for robotic applications)
    depth_stabilization:        1                               # `0`: disabled, `1`: enabled
    openni_depth_mode:          false                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters
    depth_downsample_factor:    1.0                             # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

pos_tracking:
    publish_tf:                 true                            # publish `odom -> base_link` TF
    publish_map_tf:             true                            # publish `map -> odom` TF
    map_frame:                  'map'
    odometry_frame:             'odom'
    area_memory_db_path:        ''
    area_memory:                true                            # Enable to detect loop closure
    floor_alignment:            false                           # Enable to automatically calculate camera/floor offset
    initial_base_pose:          [0.0,0.0,0.0, 0.0,0.0,0.0]      # Initial position of the `base_frame` -> [X, Y, Z, R, P, Y]
    init_odom_with_first_valid_pose: true                       # Enable to initialize the odometry with the first valid pose
    path_pub_rate:              2.0                             # Camera trajectory publishing frequency
    path_max_count:             -1                              # use '-1' for unlimited path size
    two_d_mode:                 false                           # Force navigation on a plane. If true the Z value will be fixed to "fixed_z_value", roll and pitch to zero
    fixed_z_value:              0.00                            # Value to be used for Z coordinate if `two_d_mode` is true

mapping:
    mapping_enabled:            false                           # True to enable mapping and fused point cloud pubblication
    resolution:                 0.05                            # maps resolution in meters [0.01f, 0.2f]
    max_mapping_range:          -1                              # maximum depth range while mapping in meters (-1 for automatic calculation) [2.0, 20.0]
    fused_pointcloud_freq:      1.0                             # frequency of the publishing of the fused colored point cloud
```

2. `zedm.yaml`

```
# params/zedm_yaml
# Parameters for Stereolabs ZED mini camera
---

general:
    camera_model:               'zedm'

depth:
    min_depth:                  0.35            # Min: 0.1, Max: 3.0 - Default 0.3 - Note: reducing this value wil require more computational power and GPU memory
    max_depth:                  10.0            # Max: 20.0

pos_tracking:
    imu_fusion:                 true            # enable/disable IMU fusion. When set to false, only the optical odometry will be used.

sensors:
    sensors_timestamp_sync:     false           # Synchronize Sensors messages timestamp with latest received frame
    publish_imu_tf:             true            # publish `IMU -> <cam_name>_left_camera_frame` TF
```

Myzhar, 2021-02-11:
Hi @precyon 
thank you for reporting this problem.
Can you please verify if both `imu` and `imu_raw` are affected?
Like you mentioned we already worked on this in the past and I want to understand if there was some kind of regression.
Thank you

precyon, 2021-02-12:
Thanks for the response @Myzhar. I literally have 15 bag files that show is error but I didn't record `imu` in any of them. And unfortunately (or may be fortunately), I just can not reproduce this on my system today. The timestamps look fine.

I'll am recording `imu` in my future bags and will continue to monitor this one.

closed 2021-05-17 by Myzhar