stereolabs/zed-ros-wrapper #671: [Question] Overwrite ROS image stream topic to run object detection on benchmark images

nyakasko, 2021-02-15:
Hey!

I have been wondering if it would be possible to run the object detection algorithm of ZED on images that I feed to the algorithm. Maybe somehow overwriting the image topic to publish my images?
I am inexperienced with ROS, that is why I am asking the feasability of this approach.

Could you please help me out if it is possible?

Thank you :)

Myzhar, 2021-02-15:
Hi @nyakasko 
I'm sorry, but this is not possible. Object Detection is performed by the ZED SDK and it accepts as input only data from a real camera from the ZED family or from an SVO file.

nyakasko, 2021-02-15:
> Hi @nyakasko
> I'm sorry, but this is not possible. Object Detection is performed by the ZED SDK and it accepts as input only data from a real camera from the ZED family or from an SVO file.

Thank you!

closed 2021-02-15 by nyakasko