stereolabs/zed-ros-wrapper #672: [ZED2][fix 629 doesn't work] echo /zed2/zed_node/imu/data nothing shows up.   [BUG]

gunny7828, 2021-02-16:
**Describe the bug**
/zed2/zed_node/imu/data
/zed2/zed_node/imu/data_raw
/zed2/zed_node/imu/mag
these topics seem to not publish any data.
I am using Jetpack 4.4 with Jetson Nano 2gb, cuda 10.2.
I am using sdk version : "ZED SDK for Jetpack 4.4 " 
I read the other issue similar to this and fix 619

**Device information:**
 - OS: ubuntu 18.04
 - CPU: [Jeston Nano 2gb]
[image]

 - ROS Wrapper version: not sure about the version but it contains the bug fix of November 2nd.

Myzhar, 2021-02-16:
Hi @gunny7828 
enabling `sensors/sensors_timestamp_sync` you are forcing the node to publish sensors data only when image/depth data are published.
For this reason to get sensors data you must also subscribe to one of the image/depth topics in order to start their acquisition.

gunny7828, 2021-02-16:
> Hi @gunny7828
> enabling `sensors/sensors_timestamp_sync` you are forcing the node to publish sensors data only when image/depth data are published.
> For this reason, to get sensors data you must also subscribe to one of the image/depth topics in order to start their acquisition.

Thank you so much for such quick reply. I will check it right now and get back to you in few minutes.

gunny7828, 2021-02-16:
Thanks it does work indeed. Thank you.

Myzhar, 2021-02-16:
Glad to hear. You are welcome

closed 2021-02-16 by Myzhar