stereolabs/zed-ros-wrapper #677: [Question]

zhhz9706, 2021-03-01:
Hi,
I am currently using ZED2 on TX2. When I subscribe to `/zed_node/odom`, the topic frequency I get is about 29Hz. I want to know if I can set which parameter to get a higher frequency of `odom` topics.
Thank You.

Myzhar, 2021-03-01:
@zhhz9706 the maximum value of the frequency of the odom topic is equal to the grabbing frequency.

zhhz9706, 2021-03-01:
> @zhhz9706奥多姆主题频率的最大值等于抓取频率。

so I change the grab frequency from 30 to 100, I can get 100Hz topic  frequency?

Myzhar, 2021-03-01:
100Hz is only available in VGA resolution, so you must change also the resolution parameter

zhhz9706, 2021-03-01:
> 100Hz is only available in VGA resolution, so you must change also the resolution parameter

Thank you  so  much!!!LoL

closed 2022-04-19 by github-actions[bot]