stereolabs/zed-ros-wrapper #678: [BUG]release 3.4 breaks existing code by renaming object detection variables, with no mention in release notes (people and vehicle)

fnilius, 2021-03-02:
Hi
In before v3.4 (#675), the service to start object detection used the fields : people and vehicle
Now the fields are renamed mc_people and mc_vehicle. (cf file zed_interfaces/srv/start_object_detection.srv )

This change, off course, breaks all existing code using those.
Even worse, it is not even documented in the "latest_changes.md" file.

I would suggest ideally to keep the previous naming, or at least add information about the change in the lastest_changes files.

Best regards
Felix

Myzhar, 2021-03-02:
@fnilius I can understand your disappointment.
The new version of the wrapper is fully compatible with ZED SDK v3.4 that introduces major changes in the Object Detection module.
Furthermore, we added Skeleton Tracking support and we had to change the name of the parameters to avoid eventual mismatching between Multiclass and Skeleton network models.
The changelog will be updated to match all the changes.

fnilius, 2021-03-02:
Thanks!
it's very easy to correct in the code, so it's not a big issue, but having it documented makes it far easier to understand where the compilation errors come from.

Myzhar, 2021-03-02:
The list of modifications has been added to the changelog:
https://github.com/stereolabs/zed-ros-wrapper/commit/3b3a075806e18f169e8b1846d4a4ea9bcd7c6c31

closed 2022-04-19 by github-actions[bot]