stereolabs/zed-ros-wrapper #679: [Question] Stop advertising specific topics

JohanKJIP, 2021-03-02:
Question: **Is it possible to stop advertising specific topics in any of the config files?** (in particular the depth topics)

We are using the ZED for a robotics task and don't want to perform depth calculations since we don't need it and it slows down the system. As I understand it, the calculations are not performed and published as long as we don't subscribe to the depth topics. However, the problem is that we need to record a rosbag with the -a flag and with this, the rosbag subscribes to those topics and the calculations are performed anyways.

The first attempt to solve the issue was to disable the depth calculations by setting the depth mode to 0 (MODE_NONE) in the parameter config file. With this, the zed node dies as soon as the rosbag recording is started:
```bash
[ZED][Grab] WARNING: Depth or confidence map view asked but depth map not computed (MODE_NONE)
[ZED][Grab] WARNING: Depth or confidence map view asked but depth map not computed (MODE_NONE)
```

The last resort is to edit the source in [this file](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp) on line 400 and below to disable the advertising. This is not an optimal solution, surely there must be a better way that I am missing.

Thanks

Myzhar, 2021-03-02:
Hi @JohanKJIP 
this is indeed wrong behavior. When setting depth mode to NONE we should not advertise all the topics related to depth calculation.
I suggest you modify the code to disable depth advertising and to be able to perform your task quickly, meanwhile we will schedule this modification for a future release.
It's an enhancement that requires many modifications and checks, so it requires a bit of time before it will be released.

JohanKJIP, 2021-03-02:
Thanks for the quick reply! That is understandable, I will modify the code for now 😄

Myzhar, 2024-12-12:
No modifications are scheduled for the ZED ROS Wrapper. Our current focus is on the ZED ROS2 Wrapper because ROS 1 will soon reach its end of life. If you're interested, feel free to submit a Pull Request with new features, which will be reviewed and potentially merged.

tomasz-lewicki, 2024-12-12:
Oh, my bad. I keep mixing up  `zed-ros2-wrapper` with `zed-ros-wrapper` repo.

Is this a feature that's possibly planned for `ros2` wrapper?

Myzhar, 2024-12-12:
It's already available in the ROS 2 wrapper if you disable the depth processing