stereolabs/zed-ros-wrapper #683: [Question] Which topics to use in a camera calibration?

prinsWindy, 2021-03-09:
Hi. I am trying to recalibrate my ZED 2 camera using Kalibr. I am following the guide on the Stereolabs website, but I'm confused by the image topic that is used there. 

When calibrating the camera, shouldn't we use raw, unrectified images instead of rectified images (which the guide uses)?

Myzhar, 2021-03-09:
If you have a good stable calibration it is not required to re-calibrate it and you can use the factory camera parameters.
If you think that your camera calibration is broken then you can use raw images and extract new camera parameters to be replaced in the camera setting file

prinsWindy, 2021-03-09:
I have the camera enclosed in an underwater housing and will therefore need to overwrite the factory calibration. 

So, to be clear: I need to use `/zed2/zed_node/right_raw/image_raw_color` (in addition to the left sensor) for my calibration?

Myzhar, 2021-03-09:
Yes, if you must perform a complete stereo calibration like in your case then you must add both raw topics:
`left_raw/image_raw_color` and `right_raw/image_raw_color`