stereolabs/zed-ros-wrapper #684: Not Able to run cmake

dineshrboson, 2021-03-10:
[image]

dineshrboson, 2021-03-10:
But I am able to run sample codes properly

Myzhar, 2021-03-10:
@dineshrboson I'm sorry, but it's not easy to understand the cause of your problem if you do not put all the available information together.
The compile error indicates that you are not using the latest version of the ZED SDK, this is the only certain information that I was able to get.

The [ZED SDK v3.4.1](https://www.stereolabs.com/developers/release/) supports CUDA up to version 11.1. 
In order to get everything working as expected, you must satisfy all the required dependencies:
- Remove CUDA 11.2
- Install CUDA 11.1 (or a previous version supported by the ZED SDK)
- Download and install ZED SDK v3.4.1 for the version of CUDA that you downloaded
- Compile the ZED Wrapper

dineshrboson, 2021-03-10:
cuda 11.1 is not supported for my pc

dineshrboson, 2021-03-10:
Checking CUDA version...
OK: Found CUDA 11.1
Detected nvidia driver 460.32, capable of running CUDA 11.2 (need 11.1)
Do you want to install the Object Detection module (recommended), cuDNN 8.0 and TensorRT 7.2 will be installed [Y/n] ?

dineshrboson, 2021-03-10:
Anyhow I will perform your steps and get back to you

closed 2021-03-10 by dineshrboson