stereolabs/zed-ros-wrapper #686: zed安装遇到问题，希望大佬们帮帮我.

xiaohuozi-code, 2021-03-11:
请问安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
报错：unable to identify your processor,make sure that it fits the requirements；
并且cametest：camera not detected

Myzhar, 2021-03-11:
Please post using the English language, it will be easier to reply to your questions.
@xiaohuozi-code you are posting in the wrong repository. This is a ROS-related issue, I moved it in the right position.

xiaohuozi-code, 2021-03-11:
> Please post using the English language, it will be easier to reply to your questions.
> @xiaohuozi-code you are posting in the wrong repository. This is a ROS-related issue, I moved it in the right position.

I'm so sorry!  As a beginner, I'm so sorry to disturb you. I'll pay more attention later.

Myzhar, 2021-03-11:
@xiaohuozi-code it seems that you are using the latest version of the ZED ROS wrapper with an old version of the ZED SDK.
Please update the ZED SDK to the latest v3.4, delete the folders "build" and "devel" in the root of your catkin workspace and issue a new compilation with `catkin_make`

xiaohuozi-code, 2021-03-11:
> 你make編譯沒成功 試用
> 
> > make -j8 l8

how can i resolve this problem?

xiaohuozi-code, 2021-03-11:
> > 你make編譯沒成功 試用
> > > make -j8 l8
> 
> 或是 試用python 進行編譯運作

i don't know how compile it,could you tell me in details?

Myzhar, 2021-03-11:
Sure, you can file the instructions step-by-step here:
https://github.com/stereolabs/zed-ros-wrapper#build-the-program
and here:
https://www.stereolabs.com/docs/ros/#installation

john-lihn, 2021-03-11:
他希望您刪除“ build”和“ devel”文件夾，然後使用catkin_make 重新建構 向細說明可以參考ROS教學http://wiki.ros.org/ROS/Tutorials/CreatingPackage

xiaohuozi-code, 2021-03-11:
> @xiaohuozi-code it seems that you are using the latest version of the ZED ROS wrapper with an old version of the ZED SDK.
> Please update the ZED SDK to the latest v3.4, delete the folders "build" and "devel" in the root of your catkin workspace and issue a new compilation with `catkin_make`

> 他希望您刪除“ build”和“ devel”文件夾，然後使用catkin_make 重新建構 詳細說明可以參考ROS教學http://wiki.ros.org/ROS/Tutorials/CreatingPackage

ok,我明白了，谢谢，我去尝试。

xiaohuozi-code, 2021-03-11:
> Sure, you can file the instructions step-by-step here:
> https://github.com/stereolabs/zed-ros-wrapper#build-the-program
> and here:
> https://www.stereolabs.com/docs/ros/#installation

THANK YOU VERY MUCH！I will follow your tips to change my version,thanks again!

xiaohuozi-code, 2021-03-11:
> @xiaohuozi-code it seems that you are using the latest version of the ZED ROS wrapper with an old version of the ZED SDK.
> Please update the ZED SDK to the latest v3.4, delete the folders "build" and "devel" in the root of your catkin workspace and issue a new compilation with `catkin_make`

YES,it works well, and after following your tips , "catkin_make" succeed! 
But when i run  "$roslaunch zed_wrapper zed.launch",the shell has another errors!
like this: 
RLException: [zed.launch] is neither a launch file in package [zed_wrapper] nor is [zed_wrapper] a launch file name
The traceback for the exception was written to the log file

Myzhar, 2021-03-11:
OK, this is an "environment problem". To be able to see the ROS nodes you must "source" the ROS scripts.
You can do that manually each time you open a new console using this command:
`$ source ./devel/setup.bash`
or you can do that automatically by adding the following two lines to the file `~/.bashrc`:
```
source /opt/ros/melodic/setup.bash
source <path_to_your_catkin_workspace>/devel/setup.bash
```
replacing `<path_to_your_catkin_workspace>` with the full path to your catkin workspace root folder

xiaohuozi-code, 2021-03-15:
> OK, this is an "environment problem". To be able to see the ROS nodes you must "source" the ROS scripts.
> You can do that manually each time you open a new console using this command:
> `$ source [...]

hello,sorry to disturb you again. Can you help me to solve a problem in run ORB-SLAM2 using ZED camera;
Following is my issues:
**$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml false**
**error:**
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.85272
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.4) /home/qupengtao/opencv-3.4.4/modules/highgui/src/window.cpp:615: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvNamedWindow'

Myzhar, 2021-03-15:
Hi @xiaohuozi-code 
the problem is not related to ZED, to the ZED Wrapper, or to ROS in general, but there is something wrong with the version of OpenCV that you are using.
The error message is self-explaining:
`The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvNamedWindow'`

zhoushuailin, 2021-03-20:
> 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
> 报错：unable to identify your processor,make sure that it fits the requirements；

请问你这个问题解决了吗？我也遇到了无法识别processor的问题

shangshanruoshui123, 2021-04-28:
I have the same problem: Unable to indentify your processor.How did you solve it
> > 安装zed，运行ZED_Dignosizic,诊断相机连接时：processor
> > 报错：unable to identify your processor,make sure that it fits the requirements；
> 
> 请问你这个问题解决了吗？我也遇到了无法识别processor的

Myzhar, 2021-04-28:
@shangshanruoshui123 can you please post using the English language? 
It will be easier for all the user to read you question and reply to it

TanJinXiao, 2021-04-28:
There is an error,when I run ./ZED_Diagnostic . 
Processor: unable to identify your processor,make sure that is fits the requirements. 

But others（zed sdk、graphics card、camera test、usb） are right.

closed 2022-04-19 by github-actions[bot]