stereolabs/zed-ros-wrapper #689: [Question] How can I define the Zed2 device to be used?

JoaoNBNeves, 2021-03-15:
I am using the Zed2 camera device (dev/video1) to output to 2 loopback devices (/dev/video9,10) so I can use it in ROS and in other application at the same time, but how can I configure ROS to use one of the loopback devices? .yaml and .launch files don't seem to have "device = /dev/videoX" anywhere. Thanks.

JoaoNBNeves, 2021-04-07:
Haven't been able to solve the issue yet.

Myzhar, 2021-04-07:
You can't. The SDK does not use the Linux device path to open the camera and the ZED wrapper uses the ZED SDK to connect

JoaoNBNeves, 2021-04-08:
Do you have any suggestion on how can I achieve this? Does the SDK have any form of saving video while using the camera for other functions like object detecting?

Myzhar, 2021-04-08:
With the ZED SDK you can save an SVO file that can be used as input for a new SDK session to test different configuration parameters using the same test set.

JoaoNBNeves, 2021-04-08:
I will investigate on that. Thank you @Myzhar.

closed 2021-04-08 by JoaoNBNeves