stereolabs/zed-ros-wrapper #690: Camera resends same frame when positional tracking started in multicamera nodelet

Pyrojambo, 2021-03-15:
**Describe the bug**
While running two ZED 2 cameras at the same time on the same nodelet, if you cause the positional tracking to start by subscribing to the pointcloud or odometry, it causes the camera to get stuck and continually send the same frame continuously. It does not publish the pointcloud or odometry messages, but sends the same camera image. Stopping the ROS node and starting it again allows the camera to run as normal until you try to subscribe to the pointcloud or odometry again.
Also tested this on a laptop with the same OS and libraries and got the same issue.

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: Jetson Xavier
 - ZED SDK Version: v3.4.2
 - ROS Wrapper version: v3.4.x

**To Reproduce**
Steps to reproduce the behavior:
1. Launch two ZED 2 cameras on the same nodelet
2. Subscribe to any camera image
3. Subscribe to the pointcloud or odometry
4. See error

Attached are the ZED config files in txt format. The camera names, node names and serial numbers are set through launch files based on those in this repo

[zed2.yaml.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6143235/zed2.yaml.txt)
[zed_common.yaml.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6143237/zed_common.yaml.txt)

Myzhar, 2021-03-15:
Hi @Pyrojambo 
can you explain better what you mean with `Launch two ZED 2 cameras on the same nodelet`.
How can you launch two cameras with the same nodelet? Do you mean with the same "nodelet manager"?
If that's the case, have you used a different "camera_name" for the two cameras?

Pyrojambo, 2021-03-15:
Sorry, my mistake, I meant the same nodelet manager.

I've used the launch files below to start them with different camera and node names

[dual_zed_nodelet.launch.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6143297/dual_zed_nodelet.launch.txt)
[zed2_nodelet.launch.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6143299/zed2_nodelet.launch.txt)

Myzhar, 2021-03-15:
Can you post here the outputs of the commands `$ rqt_graph` and `$ roswtf` executed when both the nodelets run?

Pyrojambo, 2021-03-16:
I've attached the rqt_graph and roswtf for before and after the bug occurs. I used RVIZ to subscribe to the front_cam image and pointcloud, but while it resends the same frame as an image it doesn't publish any more pointclouds so that connection doesn't show in rqt_graph

[dual_zed_nodelets.pdf](https://github.com/stereolabs/zed-ros-wrapper/files/6148221/dual_zed_nodelets.pdf)
[dual_zed_nodelets_roswtf.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6148223/dual_zed_nodelets_roswtf.txt)
[dual_zed_nodelets_rviz.pdf](https://github.com/stereolabs/zed-ros-wrapper/files/6148224/dual_zed_nodelets_rviz.pdf)

Myzhar, 2021-03-16:
Ok, there is something strange: `WARNING No tf messages`
I've seen that you disabled TF publishing in the launch files, but is there any node publishing the `base_link` TF frame?
The ZED Nodelet requires a valid `base_link` to correctly calculate the frame positions

Pyrojambo, 2021-03-16:
These nodelets usually run as part of a robot that has a robot_state_publisher and uses ekf_localisation to publish the odom -> base_link. The bug occurs in both situations, regardless of whether base_link has a parent

Myzhar, 2021-03-16:
Please try to enable TF publishing and start the ZED nodes and paste here the console output, in this way I can understand better what is happening

Pyrojambo, 2021-03-16:
Here's the console output of the launch, lines 449 to the end is when the pointcloud is subscribed to
[dual_zed_nodelets_with_tf.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6148541/dual_zed_nodelets_with_tf.txt)

Pyrojambo, 2021-03-30:
Has there been any progress on this? If you need more information, please let me know

Myzhar, 2021-03-31:
Hi @Pyrojambo 
the wrapper "says" that it's waiting for "static transformation". This means that the TF tree is not well defined.
Each camera requires a correct TF structure:
`map -> odom -> base_link -> camera_center`
maybe that there is something wrong with your configuration.

In the console output I read this:
```
 * /front_cam/pos_tracking/publish_map_tf: False
 * /front_cam/pos_tracking/publish_tf: True
 ```
 this means that the TF `map -> odom` is not available if there are not other nodes that broadcast it.
 
 If you use only the two zed nodelets, and nothing other, you must configure one of them (only one) in order to publish odom and map TFs.

Pyrojambo, 2021-03-31:
I tried with one of the cameras publishing both the odom and map tfs but had the same issue. Here's the console log where we subscribe to the pointcloud at line 451
[dual_ros_nodelets_with_map_tf.txt](https://github.com/stereolabs/zed-ros-wrapper/files/6236767/dual_ros_nodelets_with_map_tf.txt)

Myzhar, 2021-05-07:
Hi @Pyrojambo 
I pushed a new branch in `zed-ros-example`: [test_multicamera](https://github.com/stereolabs/zed-ros-examples/tree/test_multicamera)
In the folder `examples/zed_multicamera_example/launch` you can find the launch file `zed_multicam_single_nodelet.launch` that creates a dual camera system with two zed_nodelet running under the same nodelet manager.

To launch the example:
`$ roslaunch zed_multicamera_example zed_multicam_single_nodelet.launch`

I created an example configuration with two ZED2 placed 35cm one to the other, setting the `base_link` in the middle:

[image]

The launch script automatically launch an RViz instance configured to correctly display the two camera models, the TF and the two pointclouds.
There is also an option to disable the RViz launch: https://github.com/stereolabs/zed-ros-examples/blob/test_multicamera/examples/zed_multicamera_example/launch/zed_multicam_single_nodelet.launch#L27

[image]

Please remember to change the serial numbers of the two cameras and the relative positions and orientations respect to `base_link`, otherwise the example will not work for your system.

The example creates a "global" namespace called `zed_multi` where the nodelet manager is placed.
The two nodelets have their own namespaces `zed2_front` and `zed2_rear` to be able to discern the services, the parameters and the TF links.

See the following rqt graph for clarity:

[image]

Please let me know if this is the configuration that you were trying to achieve and do not hesitate to ask for details if you need.

Pyrojambo, 2021-05-17:
Hi @Myzhar,
We tested your launch file with the different namespaces but still had the same error.
Did you test this on a Nvidia Jetson Xavier or something else?

Myzhar, 2021-05-17:
@Pyrojambo I tested the branch on the Xavier NX and it works as expected, even if I noticed a few things to check:
1) rviz or any other high demanding node cannot run on the Xavier NX itself but they must be hosted by other machines
2) the `map`->`odom`->`base_link` TF chain is published only when subscribing to point cloud, to pose, or to odometry topics of the "main" camera
3) sometimes the Xavier NX stops working and emits the warning "System throttled due to Over-current"

For the first point, there is nothing to do. Xavier NX is powerful, but it's not a desktop PC.
For the second point, we must work on a modification to the behavior of the ZED Wrapper to start the positional tracking issue automatically when the node starts if requested by user parameters (the same as Object detection, mapping and other services)
For the third and last point, I suppose that an externally powered UB3 hub can help, but I have not tested it yet.

Myzhar, 2021-05-17:
@Pyrojambo I just published an update with an important fix about this issue in the new branch `test_multicamera` of the ZED Wrapper repository:
https://github.com/stereolabs/zed-ros-wrapper/tree/test_multicamera
Can I ask you to test it to verify that it solves your problems with TF broadcasting?

Pyrojambo, 2021-07-01:
We are running this on an Nvidia AGX Xavier so don't have any power or computation issues that you get on the Xavier NX.

But we finally tested this properly again and the changes you made have fixed our issue.
Thank you kindly

closed 2021-07-01 by Pyrojambo